swri-robotics/bag-database #22: Get a direct download link for a given bag

ugocupcic, 2016-05-11:
It'd be super useful to be able to get a direct download link (e.g. to use in a test with wget) from the webui!

pjreed, 2016-05-11:
Hmm... I was about to say that you can right-click on the "Save" icon and copy the link, but I just realized that when I added the context menu, that prevents the normal menu from appearing, so that doesn't quite work any more. 

Maybe I'll add a "Copy Link" option to the context menu; would that work?

In the meantime, you can manually craft a URL for downloading a bag like `http://bag-db:8080/bags/download?bagId=1176`, where the number on the end is the database id of the bag.

ugocupcic, 2016-05-11:
Thanks for the quick reply @pjreed ! The "Copy Link" would be perfect yes.

Thanks for the crafty url too.

PR #23 (2016-05-17): Adding the ability to copy links to download bags
Resolves #22
commit 33b31709 (2016-05-17): Adding the ability to copy links to download bags

Resolves #22
closed 2016-05-17 by pjreed