swri-robotics/bag-database #26: Show video streams for image topics

pjreed, 2016-05-17:
Currently it's possible to display the first image on a topic, but it'd be really handy to be able to stream the topic as a video.

fabolhak, 2016-06-19:
I'm wondering if it is possible to create this videos on demand (when the user clicks on the icon) or if they should be preprocessed. Unfortunately my Java experience is very little. So I can't tell what is feasible.

Additionally we made some good experience with "time-lapsing" the videos (e.g. showing 1 frame per second). This gives the user a faster overview what’s going on in the bag. Maybe the time lapse effect can be controlled via a parameter?

pjreed, 2016-06-20:
How feasible it is to create videos on demand may depend on the resolution and frame rate of the image topic.  I think that at least 720p at 30 fps should be possible on any modern CPU.  Random access of data in bag files is very slow, but reading data sequentially is pretty fast, so it's just a matter of figuring out how to encode the images into a video stream and return it to the user.  I'm not familiar with any Java video encoding libraries, so it's just a matter of me finding a good one and figuring out how to use it.

PR #55 (2017-01-12): Add ability to view video streams of image topics
This adds two new icons to the Topics grid in the Bag Details window.
Both of them will open video streams for supported message types, which include:
- sensor_msgs/Image
- sensor_msgs/CompressedImage
- stereo_msgs/DisparityImage

Messages from the topic will be converted to a VP8 video stream using ffmpeg
and streamed back to the user.  The second icon allows you to create a video
stream that skips every n images, effectively fast-forwarding through the stream.

Resolves #26
commit 8c149105 (2017-01-12): Add ability to view video streams of image topics (#55)

Add ability to view video streams of image topics

This adds two new icons to the Topics grid in the Bag Details window.
Both of them will open video streams for supported message types, which include:

- sensor_msgs/Image
- sensor_msgs/CompressedImage
- stereo_msgs/DisparityImage

Messages from the topic will be converted to a VP8 video stream using ffmpeg
and streamed back to the user.  The second icon allows you to create a video
stream that skips every n images, effectively fast-forwarding through the stream.

Resolves #26
closed 2017-01-12 by pjreed