swri-robotics/bag-database #27: No metadata identified for some bag files

beatrizleon, 2016-06-06:
I uploaded to the server folder a few big bag files (around 4 GB each) and although they appear correcltly in the list, the message types and topics are empty. Do you know what could be the issue? 

Here are some screenshots showing the problem:
[image]

[image]

pjreed, 2016-06-07:
On the main page, there will be an "# Errors" button in the lower left if it had problems parsing bag files; if that indicates there were any errors, could you click on that and let me know what they are?  Alternately, check the log file (`docker logs bagdb` if you're using docker) and see if it's printed out any exceptions.

If neither of those things are helpful, could you use `rosbag` to filter one of your bag files into a smaller file, test to see whether that has the same issue, and send it to me (via Dropbox or something similar) to see if I can reproduce the problem?

ugocupcic, 2016-06-08:
Hi @pjreed, I'm working with @beatrizleon. There's no error on the main page (0 errors). I couldn't spot anything obvious [in that log](https://gist.github.com/ugocupcic/129659b2f932192c4b6950524c78e136). Cheers

pjreed, 2016-06-08:
The log file looks like it's parsing message types out of bags and inserting them properly; are other bags displaying properly?  Also, if you look in your browser's error console (hit F12) when displaying the details for a bag file, does it show any errors?

ugocupcic, 2016-06-08:
Yes the other bags are OK! Only those very big bag files are not.

Couldn't spot any errors with the browser's console:

```
Opening Web Socket...
stomp.min.js:8 Web Socket Opened...
stomp.min.js:8 >>> CONNECT
accept-version:1.1,1.0
heart-beat:10000,10000

 
stomp.min.js:8 <<< CONNECTED
version:1.1
heart-beat:0,0

 
stomp.min.js:8 connected to server undefined
stomp.min.js:8 >>> SUBSCRIBE
id:sub-0
destination:/topic/status
```

pjreed, 2016-06-08:
Hmm, that looks fine.  We've got well over 100 bags in our database that are over 4 GB (the largest is over 100 GB), so I don't think the size should be the issue.

I've got a couple more ideas...
- If you run `rosbag reindex` on the bag, is the reindexed bag different from the original?  Does the Bag DB have the same behavior when displaying its images?
- Could you make a new file that's a smaller slice of the original (see http://wiki.ros.org/rosbag/Cookbook for some examples) and see if that has the same problem?  If so, could you send that bag to me?

beatrizleon, 2016-06-14:
I run the reindex cmd but the new file had no difference with the original. Also tried to upload it but still no infomation is displayed. I got an error:
[image]

beatrizleon, 2016-06-14:
Cropped file is fine:
[image]
Maybe it is something due to the transmition? In any case, when we download them they work fine so it is not a big issue just something nice to have.

pjreed, 2016-06-14:
How strange!   Are there any topics or message types in the full bag file that aren't in the cropped one?

beatrizleon, 2016-06-14:
no :)

pjreed, 2016-06-14:
I wonder if it might be an issue with the number of messages rather than the file size; could you run `rosbag info` on the original and let me know how many messages it says are on each topic?

beatrizleon, 2016-06-14:
[image]

pjreed, 2016-06-14:
That doesn't seem like an unusually larger number of messages, I know I've processed some that have several dozen thousand on a topic... however, there are two things I notice:
- The bag is bzipped, and in theory support for compressed bags works and is implemented, we don't have many compressed bag files, so I haven't tested it thoroughly.
- Those images are much larger than ones I've tested with; I've got one uncompressed bag file with about 13k images that is 12 GB, so only about 50% larger than your file that has 1/6 as many images.  Maybe message size could factor into it.

I'll see if I can construct a few bag files to test things out.

pjreed, 2016-06-17:
Well, I've spent a while trying to reproduce it with various combinations of bag files with CameraInfos and Images, and I still haven't had any luck.  Do you have any way that you could share the original bag file with me?

fabolhak, 2016-06-19:
Hello @pjreed , 
I also tried the ROSBag Database yesterday. 
First of all I worked pretty good for our system, thank you for contributing this project :+1: ! 

However I had some similar problems: When copying bag files to the /location/bags folder **while** the docker image was **running**, the Bag Database had problems getting the metadata for some bags (around 1 out of 3 bags). 

When I stop the docker image. Copy the bag files first and restart the docker image after all the copying has finished, the problem disappeared.

It seems like the Bag Database tries to get the metadata from Bagfiles which are not finished copying. Could this be the issue?

Tomorrow when I'm back in the office, I can provide more detailed infos and investigate a little bit more.

Kindest Regards

pjreed, 2016-06-20:
That's a good idea.  I've only been testing on very fast disk arrays, so even bags that are several GB have finished writing by the time the Bag DB tried to read them.  I'll test large files on slower disks and see if I have any problems.

pjreed, 2016-06-20:
Ok, I think I've managed to reproduce the problem.

When listening for filesystem events, the Bag DB actually already waits 3 seconds for events to stop before it begins reading files, and in theory that would stop it from trying to read a file that was still being written.  It looks like the problem happens when the filesystem settles, it begins reading bags, and then another bag is appears before it finishes reading the earlier ones; it can begin reading the second bag before it's done being written, and then it will have an error and mark the file as not having any messages in it.

A few other symptoms of the issue:
- In the bag details, it is listed as having 0 messages
- The file size in the details is less than the file's actual size
- The start and end times are both 1/1/1970

I've got a few different ideas for a fix, I'll try them and see what works well.

pjreed, 2016-06-20:
I've released a new version that I believe will fix this issue.  Let me know if you still have the same problem in version 2.3.

For databases that currently have listing without metadata, there are two ways to fix it:
1. Drop the database and recreate it from scratch
or, if you want to preserve an existing database:
1. Move the afflicted bag files out of the DB's directory so that it can no longer see them
2. Wait a minute, then reload the bag and verify that the bag DB thinks they're missing (the entries will be italicized)
3. Log in, then open the Administration panel and click `Remove DB entries for missing bags`
4. Wait until it's done
5. Move the bag files back into the DB's directory

Note that there have been some significant changes in the database schema in this version; the upgrade should happen automatically, but you might want to make a backup beforehand, and let me know if anything doesn't work right.

fabolhak, 2016-06-21:
This works for us! Every bag file has now full meta data even when the docker container is running and the files are still copying. Thank you very much :+1: !

ugocupcic, 2016-06-23:
I just updated the docker image here and it's working as well for us. Thanks for the quick fix!!

beatrizleon, 2016-06-23:
Thanks @pjreed for fixing the issue! I guess we can close it now

closed 2016-06-23 by beatrizleon