swri-robotics/bag-database #32: Show folders

fabolhak, 2016-06-20:
Hello,
we are storing our bag files in a folder structure for a better overview. The bag file names contain only the date and time (hard to differentiate). I was wondering if there are any plans to implement a 'folder view'  in the ROS Bag database as well. This would be very handy for us. For example:

- Scene1
  - Measurement 1
     - file1.bag
     - file2.bag
  - Measurement 2
    - file1.bag
- Scene2
   - Measurement 1
     - file1.bag

....

pjreed, 2016-06-20:
This could be useful to have.  One concern is that a tree view may not render as quickly for thousands of bags as the current list view does, but I'll investigate and see if it's feasible.

In the mean time, it is hidden by default, but you can add a "Path" column to the list that will display the full path to the file:
[image]
Currently the state of the menus is not saved when you reload, so you would have to add the column every time you go to the page, and that's a little annoying.  I can also look into saving the menu state if that would be useful.

pjreed, 2016-06-22:
I've been working on adding a folder view; I haven't made an official release yet, but if you'd like you can test it out by using the `swrirobotics/bag-database:2.4-SNAPSHOT` docker image.  Let me know what you think!
[image]
One thing to note is that the folder view does not support searching through text fields and metadata in the database like the list view does; it only supports filtering the view by file name.

fabolhak, 2016-06-22:
Hi,
thank you for your quick reply. The snapshot works on my system :). I think the folder View is pretty cool giving a nice overview, when the user has a more sophisticated folder structure. I originally thought of a combined view of "List View" and "Folder View", but as you already mentioned this is probably not so easy to implement. I guess this also does the job very well. Thank you again :+1:

PR #35 (2016-06-21): Implementing a hierarchical folder view for bags.
Resolves #32
commit 7d52dc30 (2016-06-21): Implementing a hierarchical folder view for bags.

Resolves #32
closed 2016-09-30 by pjreed