swri-robotics/bag-database #43: Read key:value pairs from metadata topic

cbandera, 2016-07-03:
Hi,

would it by possible to add the functionality to read several key:value pairs from a specified topic?
I think of using a std::msgs/String message similar to the vehicle_name topic. But instead of providing just a single word, I would like to pass a whole dict of key:value pairs, possibly in yaml format:
```yaml
General:
  CreatedBy: user
  Description: "A drive through the park"
Environment:
  Location: Some Park
  Weather: Sunny
Platform:
  VehicleName: ...
  Sensors: Cameras...
```
This schema could even be assumed fixed, through e.g. a config file. I don't have enough experience with java to be able to fully understand how the database is build up. Could someone give me an estimate of how complicated that would be? (e.g. is it possible with the way the DB is build up right now)

The advantage of having such a functionality is, that you could insert this metadata into the bagfile at record time (through additional tooling, like [rosbag_metadata](https://github.com/hordurk/rosbag_metadata)). Forcing the user to specify data at the time of recording, and placing the metadata in the same file as the recording itself has several advantages, e.g. (you could access this information directly from the console, you don't have to rely on people providing the metainformation afterwards, when uploading the file to the database...)

pjreed, 2016-07-05:
This seems like it could be useful, although currently there's no way to store arbitrary metadata about a bag in the database.  There is a `tags` table that was created with the intent of being able to tag bag files with simple strings that could later be used to search for them (i. e., "test", "failure", "demo", etc.), although no interface for that is currently implemented in the GUI.

In order to add something like this, I think the work would involve:
- Design a table for storing the data; sets of key/value pairs would be simple, but arbitrary yaml is much more complex
- Create a liquibase changelog to create the new table 
- Implement a persistence class that represents the table and its associations with other tables
- Modify the BagScanner to read the data from the bag files
- Modify the BagService to save the data to the database

We would probably also want to be able to configure which topic it reads this information from, which would involve:
- Add a new field to the Configuration class
- Add a new field to the ConfigWindow.js view

Of course, just being able to read and store the information on the backend isn't useful if there's no way for a user to be able to search for or display it.  Also doing that will involve some more work:
- Add a new model for storing the metadata
- Design a way to display the metadata to the user, which will probably involve modifying the BagGrid.js and/or the BagDetailsWindow.js views
- Add an option to the SearchPanel.js view for searching metadata
- Modify BagService to handle filtering by and searching for the metadata

All in all, I don't think it is particular *complicated* (well, for somebody who is familiar with Spring/Hibernate/ExtJS), but it is a fair amount of work.

I've looked at rosbag_metadata before and I think adding support for reading data with that would be useful; I would actually prefer that to storing the data on a topic in the bag file.  It's reasonable that you might want to edit a bag's metadata after it's being loaded into the database (or save changes made through the DB's interface), and modifying the actual bag file would cause the DB to think it is a different bag from the original, which could cause some complications.  I think the work necessary for that would be mostly the same, except the BagScanner would need to find and read the associated metadata file, and you would probably want to add a way to the Admin panel to force re-scanning metadata.

pjreed, 2017-01-12:
Implemented in #60.

closed 2017-01-12 by pjreed