swri-robotics/bag-database #49: The file still exists in explorer after the file is deleted from the volume folder

tngan, 2016-09-29:
I put 3 bags in the mounted volume (default is /bag/location), the explorer shows properly. However, when I delete one of them and refresh the explorer, all 3 bags are still there even though the remote folder (/data) inside the container remains 2 files.

pjreed, 2016-09-30:
Currently, if a bag has disappeared from the file system, it won't be removed from the database, but instead its text will be italicized to indicate that it's missing.  Missing entries can be removed from the database with the "Remove DB Entries for Missing Bags" button in the Administration panel.

I originally did this because I thought that, since users can enter custom metadata for bag files, users might not want those entries to automatically disappear just because the file is gone.  Do you think it'd make sense to automatically remove entries?  (or maybe it would be better to have a config option for it?)

tngan, 2016-09-30:
@pjreed Thanks for clarification. Basically, I do think the entries should be deleted if the file is gone, users should understand what they are doing. Or, if we want to keep the history, we may want to archive the records instead of delete everything.

PR #52 (2016-10-14): Add ability to remove bags from DB when deleted from filesystem
This adds a new config value, "removeOnDeletion", that will cause the bag DB
to remove entries from the database if the associated files disappeared from
the filesystem.  This will be set to true by default.

It also makes a few tweaks to the database schema so that foreign keys associated with the `bags` table all have "ON DELETE CASCADE" set on them, so that when bags are removed, the database will remove related rows; this is much faster than using Hibernate to delete them manually.

Resolves #49
commit 0074581d (2016-10-14): Add ability to remove bags from DB when deleted from filesystem (#52)

This adds a new config value, "removeOnDeletion", that will cause the bag DB
to remove entries from the database if the associated files disappeared from
the filesystem.  This will be set to true by default.

Resolves #49
closed 2017-01-12 by pjreed