swri-robotics/bag-database #66: Docker images not working out of the box

Timple, 2017-03-08:
Although I am aware that you are using this internally, we are starting to look into storing bagfiles and this solution looked quite promising.

However the docker files do not seem to work on my PC. These are the two docker containers I am spawning locally:

```docker run -d     --name bagdb-postgres     -v /var/lib/bagdb-postgres:/var/lib/postgresql/data     -e POSTGRES_PASSWORD=letmein     -e POSTGRES_USER=bag_database     -e POSTGRES_DB=bag_database     mdillon/postgis:latest```

```docker run -d     -p 8888:8080     -v /home/tim/rosbag:/bags     --name bagdb     --link bagdb-postgres:bagdb-postgres     -e DB_DRIVER=org.postgresql.Driver     -e DB_PASS=letmein     -e DB_URL="jdbc:postgresql://bagdb-postgres/bag_database"     -e DB_USER=bag_database     -e METADATA_TOPICS="/metadata"     -e VEHICLE_NAME_TOPICS="/vehicle_name"     -e GPS_TOPICS="/dory/gps/fix, /phone1/android/gps, /imu/fix"     swrirobotics/bag-database:latest```

And this is the resulting log files:

```
mkdir: cannot create directory ‘/root/.ros-bag-database’: File exists
08-Mar-2017 10:18:44.060 WARNING [main] org.apache.catalina.startup.SetAllPropertiesRule.begin [SetAllPropertiesRule]{Server/Service/Connector} Setting property 'maxSpareThreads' to '75' did not find a matching property.
08-Mar-2017 10:18:44.111 INFO [main] org.apache.catalina.startup.VersionLoggerListener.log Server version:        Apache Tomcat/8.0.37
08-Mar-2017 10:18:44.111 INFO [main] org.apache.catalina.startup.VersionLoggerListener.log Server built:          Sep 1 2016 10:01:52 UTC
08-Mar-2017 10:18:44.111 INFO [main] org.apache.catalina.startup.VersionLoggerListener.log Server number:         8.0.37.0
08-Mar-2017 10:18:44.111 INFO [main] org.apache.catalina.startup.VersionLoggerListener.log OS Name:               Linux
08-Mar-2017 10:18:44.112 INFO [main] org.apache.catalina.startup.VersionLoggerListener.log OS Version:            4.8.0-34-generic
08-Mar-2017 10:18:44.112 INFO [main] org.apache.catalina.startup.VersionLoggerListener.log Architecture:          amd64
08-Mar-2017 10:18:44.112 INFO [main] org.apache.catalina.startup.VersionLoggerListener.log Java Home:             /usr/lib/jvm/java-8-openjdk-amd64/jre
08-Mar-2017 10:18:44.112 INFO [main] org.apache.catalina.startup.VersionLoggerListener.log JVM Version:           1.8.0_102-8u102-b14.1-1~bpo8+1-b14
08-Mar-2017 10:18:44.112 INFO [main] org.apache.catalina.startup.VersionLoggerListener.log JVM Vendor:            Oracle Corporation
08-Mar-2017 10:18:44.113 INFO [main] org.apache.catalina.startup.VersionLoggerListener.log CATALINA_BASE:         /usr/local/tomcat
08-Mar-2017 10:18:44.113 INFO [main] org.apache.catalina.startup.VersionLoggerListener.log CATALINA_HOME:         /usr/local/tomcat
08-Mar-2017 10:18:44.114 INFO [main] org.apache.catalina.startup.VersionLoggerListener.log Command line argument: -Djava.util.logging.config.file=/usr/local/tomcat/conf/logging.properties
08-Mar-2017 10:18:44.114 INFO [main] org.apache.catalina.startup.VersionLoggerListener.log Command line argument: -Djava.util.logging.manager=org.apache.juli.ClassLoaderLogManager
08-Mar-2017 10:18:44.115 INFO [main] org.apache.catalina.startup.VersionLoggerListener.log Command line argument: -Djdk.tls.ephemeralDHKeySize=2048
08-Mar-2017 10:18:44.115 INFO [main] org.apache.catalina.startup.VersionLoggerListener.log Command line argument: -Djava.endorsed.dirs=/usr/local/tomcat/endorsed
08-Mar-2017 10:18:44.115 INFO [main] org.apache.catalina.startup.VersionLoggerListener.log Command line argument: -Dcatalina.base=/usr/local/tomcat
08-Mar-2017 10:18:44.115 INFO [main] org.apache.catalina.startup.VersionLoggerListener.log Command line argument: -Dcatalina.home=/usr/local/tomcat
08-Mar-2017 10:18:44.116 INFO [main] org.apache.catalina.startup.VersionLoggerListener.log Command line argument: -Djava.io.tmpdir=/usr/local/tomcat/temp
08-Mar-2017 10:18:44.116 INFO [main] org.apache.catalina.core.AprLifecycleListener.lifecycleEvent Loaded APR based Apache Tomcat Native library 1.2.8 using APR version 1.5.1.
08-Mar-2017 10:18:44.117 INFO [main] org.apache.catalina.core.AprLifecycleListener.lifecycleEvent APR capabilities: IPv6 [true], sendfile [true], accept filters [false], random [true].
08-Mar-2017 10:18:44.123 INFO [main] org.apache.catalina.core.AprLifecycleListener.initializeSSL OpenSSL successfully initialized (OpenSSL 1.0.2i  22 Sep 2016)
08-Mar-2017 10:18:44.198 INFO [main] org.apache.coyote.AbstractProtocol.init Initializing ProtocolHandler ["http-apr-8080"]
08-Mar-2017 10:18:44.216 INFO [main] org.apache.catalina.startup.Catalina.load Initialization processed in 480 ms
08-Mar-2017 10:18:44.239 INFO [main] org.apache.catalina.core.StandardService.startInternal Starting service Catalina
08-Mar-2017 10:18:44.239 INFO [main] org.apache.catalina.core.StandardEngine.startInternal Starting Servlet Engine: Apache Tomcat/8.0.37
08-Mar-2017 10:18:44.264 INFO [localhost-startStop-1] org.apache.catalina.startup.HostConfig.deployWAR Deploying web application archive /usr/local/tomcat/webapps/ROOT.war
08-Mar-2017 10:18:47.390 INFO [localhost-startStop-1] org.apache.jasper.servlet.TldScanner.scanJars At least one JAR was scanned for TLDs yet contained no TLDs. Enable debug logging for this logger for a complete list of JARs that were scanned but no TLDs were found in them. Skipping unneeded JARs during scanning can improve startup time and JSP compilation time.
2017-03-08 10:18:47.874 [localhost-startStop-1] INFO  o.s.web.context.ContextLoader - Root WebApplicationContext: initialization started
2017-03-08 10:18:47.898 [localhost-startStop-1] INFO  o.s.w.c.s.AnnotationConfigWebApplicationContext - Refreshing Root WebApplicationContext: startup date [Wed Mar 08 10:18:47 UTC 2017]; root of context hierarchy
2017-03-08 10:18:47.995 [localhost-startStop-1] INFO  o.s.w.c.s.AnnotationConfigWebApplicationContext - Registering annotated classes: [class com.github.swrirobotics.config.ApplicationConfig,class com.github.swrirobotics.config.JpaConfig,class com.github.swrirobotics.config.LiquibaseConfig,class com.github.swrirobotics.config.WebSocketConfig,class com.github.swrirobotics.config.SecurityConfig]
2017-03-08 10:18:49.500 [localhost-startStop-1] INFO  o.s.b.f.a.AutowiredAnnotationBeanPostProcessor - JSR-330 'javax.inject.Inject' annotation found and supported for autowiring
2017-03-08 10:18:49.606 [localhost-startStop-1] INFO  c.g.s.config.JpaConfig - entityManagerFactory
2017-03-08 10:18:49.692 [localhost-startStop-1] INFO  c.g.s.config.JpaConfig - JDBC driver: org.postgresql.Driver
2017-03-08 10:18:49.696 [localhost-startStop-1] INFO  c.g.s.config.JpaConfig - JDBC URL: jdbc:postgresql://bagdb-postgres/bag_database
2017-03-08 10:18:49.698 [localhost-startStop-1] INFO  c.g.s.config.JpaConfig - JDBC Username: bag_database
2017-03-08 10:18:49.730 [localhost-startStop-1] INFO  c.zaxxer.hikari.HikariDataSource - HikariPool-1 - Started.
2017-03-08 10:18:52.860 [localhost-startStop-1] WARN  o.s.w.c.s.AnnotationConfigWebApplicationContext - Exception encountered during context initialization - cancelling refresh attempt: org.springframework.beans.factory.BeanCreationException: Error creating bean with name 'entityManagerFactory' defined in com.github.swrirobotics.config.JpaConfig: Bean instantiation via factory method failed; nested exception is org.springframework.beans.BeanInstantiationException: Failed to instantiate [org.springframework.orm.jpa.LocalContainerEntityManagerFactoryBean]: Factory method 'entityManagerFactory' threw exception; nested exception is org.springframework.beans.factory.BeanCreationException: Error creating bean with name 'dataSource' defined in com.github.swrirobotics.config.JpaConfig: Bean instantiation via factory method failed; nested exception is org.springframework.beans.BeanInstantiationException: Failed to instantiate [javax.sql.DataSource]: Factory method 'dataSource' threw exception; nested exception is com.zaxxer.hikari.pool.HikariPool$PoolInitializationException: Failed to initialize pool: The connection attempt failed.
2017-03-08 10:18:52.863 [localhost-startStop-1] ERROR o.s.web.context.ContextLoader - Context initialization failed
org.springframework.beans.factory.BeanCreationException: Error creating bean with name 'entityManagerFactory' defined in com.github.swrirobotics.config.JpaConfig: Bean instantiation via factory method failed; nested exception is org.springframework.beans.BeanInstantiationException: Failed to instantiate [org.springframewo [... cut ...]
 java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
 java.lang.Thread.run(Thread.java:745)
08-Mar-2017 10:18:52.893 INFO [localhost-startStop-1] org.apache.catalina.startup.HostConfig.deployWAR Deployment of web application archive /usr/local/tomcat/webapps/ROOT.war has finished in 8,629 ms
08-Mar-2017 10:18:52.894 INFO [main] org.apache.coyote.AbstractProtocol.start Starting ProtocolHandler ["http-apr-8080"]
08-Mar-2017 10:18:52.898 INFO [main] org.apache.catalina.startup.Catalina.start Server startup in 8681 ms
```

pjreed, 2017-03-08:
It looks like the bag-database container is failing to resolve the hostname for the postgres container.  That's a little odd, as the commands I've used to start it locally are almost identical, and I don't see anything obviously wrong with yours.  What's your host OS and what version of docker are you using?  (`docker -v`)

Something you might try is creating a docker network and placing both containers on the network instead of using the `--link` option.  The steps for doing that would look something like:

Create the network:
```bash
sudo docker network create bagdb
```

Create the postgres container:
```bash
sudo docker run -d \
    --net bagdb \
    --name bagdb-postgres \
    -v /var/lib/bagdb-postgres:/var/lib/postgresql/data \
    -e POSTGRES_PASSWORD=letmein \
    -e POSTGRES_USER=bag_database \
    -e POSTGRES_DB=bag_database \
    mdillon/postgis:latest
```

Create the Bag DB container:
```
docker run -d \
    -p 8888:8080 \
    -v /home/tim/rosbag:/bags \
    --name bagdb \
    --net bagdb \
    -e DB_DRIVER=org.postgresql.Driver \
    -e DB_PASS=letmein \
    -e DB_URL="jdbc:postgresql://bagdb-postgres/bag_database" \
    -e DB_USER=bag_database \
    -e METADATA_TOPICS="/metadata" \
    -e VEHICLE_NAME_TOPICS="/vehicle_name" \
    -e GPS_TOPICS="/dory/gps/fix, /phone1/android/gps, /imu/fix" \
    swrirobotics/bag-database:latest
```

Timple, 2017-03-08:
That would be:

```
uname -a
Linux Precision 4.8.0-34-generic #36~16.04.1-Ubuntu SMP Wed Dec 21 18:55:08 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
```
```
docker -v
Docker version 17.03.0-ce, build 60ccb22
```
(Latest from the `deb https://apt.dockerproject.org/repo ubuntu-xenial main` repository)

I will try your proposed solution Thursday

pjreed, 2017-03-29:
Out of curiosity, did you have the chance to give it another try?

Timple, 2017-03-30:
I did just now :). It fell off my radar for a little bit.

I still get the `mkdir: cannot create directory ‘/root/.ros-bag-database’: File exists` and also the 

```
2017-03-30 07:12:31.773 [localhost-startStop-1] INFO  o.s.w.c.s.AnnotationConfigWebApplicationContext - Registering annotated classes: [class com.github.swrirobotics.config.WebMvcConfig]
2017-03-30 07:12:31.859 [localhost-startStop-1] INFO  o.s.b.f.a.AutowiredAnnotationBeanPostProcessor - JSR-330 'javax.inject.Inject' annotation found and supported for autowiring
2017-03-30 07:12:31.915 [localhost-startStop-1] DEBUG c.g.s.account.AccountRepository - Looking up user by email: admin
2017-03-30 07:12:31.920 [localhost-startStop-1] WARN  c.g.s.account.AccountRepository - No user found
org.springframework.dao.EmptyResultDataAccessException: No entity found for query; nested exception is javax.persistence.NoResultException: No entity found for query
	at org.springframework.orm.jpa.EntityManagerFactoryUtils.convertJpaAccessExceptionIfPossible(EntityManagerFactoryUtils.java:392)
	at org.springframework.orm.jpa.aspectj.JpaExceptionTranslatorAspect.ajc$afterThrowing$org_springframework_orm_jpa_aspectj_JpaExceptionTranslatorAspect$1$18a1ac9(JpaExceptionTranslatorAspect.aj:37)
	at com.github.swrirobotics.account.AccountRepository.findByEmail_aroundBody2(AccountRepository.java:69)
	at com.github.swrirobotics.account.AccountRepository$AjcClosure3.run(AccountRepository.java:1)
	at org.springframework.transaction.aspectj.AbstractTransactionAspect.ajc$around$org_springframework_transaction_aspectj_AbstractTransactionAspect$1$2a73e96cproceed(AbstractTransactionAspect.aj:66)
	at org.springframework.transaction.aspectj.AbstractTransactionAspect$AbstractTransactionAspect$1.proceedWithInvocation(AbstractTransactionAspect.aj:72)
	at org.springframework.transaction.interceptor.TransactionAspectSupport.invokeWithinTransaction(TransactionAspectSupport.java:281)
	at org.springframework.transaction.aspectj.AbstractTransactionAspect.ajc$around$org_springframework_transaction_aspectj_AbstractTransactionAspect$1$2a73e96c(AbstractTransactionAspect.aj:70)
	at com.github.swrirobotics.account.AccountRepository.findByEmail(AccountRepository.java:64)
	at com.github.swrirobotics.bags.admin.AdminController$1.doInTransactionWithoutResult(AdminController.java:88)
	at org.springframework.transaction.support.TransactionCallbackWithoutResult.doInTransaction(TransactionCallbackWithoutResult.java:34)
	at org.springframework.transaction.support.TransactionTemplate.execute(TransactionTemplate.java:133)
	at com.github.swrirobotics.bags.admin.AdminController.checkAdminAccount(AdminController.java:85)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.springframework.beans.factory.annotation.InitDestroyAnnotationBeanPostProcessor$LifecycleElement.invoke(InitDestroyAnnotationBeanPostProcessor.java:365)
	at org.springframework.beans.factory.annotation.InitDestroyAnnotationBeanPostProcessor$LifecycleMetadata.invokeInitMethods(InitDestroyAnnotationBeanPostProcessor.java:310)
	at org.springframework.beans.factory.annotation.InitDestroyAnnotationBeanPostProcessor.postProcessBeforeInitialization(InitDestroyAnnotationBeanPostProcessor.java:133)
	at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.applyBeanPostProcessorsBeforeInitialization(AbstractAutowireCapableBeanFactory.java:408)
	at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.initializeBean(AbstractAutowireCapableBeanFactory.java:1575)
	at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.doCreateBean(AbstractAutowireCapableBeanFactory.java:545)
	at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.createBean(AbstractAutowireCapableBeanFactory.java:482)
	at org.springframework.beans.factory.support.AbstractBeanFactory$1.getObject(AbstractBeanFactory.java:306)
	at org.springframework.beans.factory.support.DefaultSingletonBeanRegistry.getSingleton(DefaultSingletonBeanRegistry.java:230)
	at org.springframework.beans.factory.support.AbstractBeanFactory.doGetBean(AbstractBeanFactory.java:302)
	at org.springframework.beans.factory.support.AbstractBeanFactory.getBean(AbstractBeanFactory.java:197)
	at org.springframework.beans.factory.support.DefaultListableBeanFactory.preInstantiateSingletons(DefaultListableBeanFactory.java:751)
	at org.springframework.context.support.AbstractApplicationContext.finishBeanFactoryInitialization(AbstractApplicationContext.java:861)
	at org.springframework.context.support.AbstractApplicationContext.refresh(AbstractApplicationContext.java:541)
	at org.springframework.web.servlet.FrameworkServlet.configureAndRefreshWebApplicationContext(FrameworkServlet.java:668)
	at org.springframework.web.servlet.FrameworkServlet.initWebApplicationContext(FrameworkServlet.java:540)
	at org.springframework.web.servlet.FrameworkServlet.initServletBean(FrameworkServlet.java:494)
	at org.springframework.web.servlet.HttpServletBean.init(HttpServletBean.java:136)
	at javax.servlet.GenericServlet.init(GenericServlet.java:158)
	at org.apache.catalina.core.StandardWrapper.initServlet(StandardWrapper.java:1227)
	at org.apache.catalina.core.StandardWrapper.load(StandardWrapper.java:1030)
[... cut ...]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)
Caused by: javax.persistence.NoResultException: No entity found for query
	at org.hibernate.query.internal.AbstractProducedQuery.getSingleResult(AbstractProducedQuery.java:1446)
	... 49 common frames omitted
2017-03-30 07:12:31.923 [localhost-startStop-1] INFO  c.g.s.bags.admin.AdminController - No admin account found; creating a default one.
```

But it does seem to be working for the rest :).

pjreed, 2017-03-30:
Those are actually normal, so no need to worry about them.  Let me know if you have any other issues!

closed 2017-03-30 by pjreed