swri-robotics/bag-database #69: Use S3 for bag storage

clecocel, 2017-04-20:
It would be nice to be able to connect the bag database to an S3 bucket that would hold the bag files instead of needing to have them on disk.

pjreed, 2017-04-26:
This would be nice to have, but will probably also involve a complete rewrite of the BagService class, and it might make sense to turn that into an interface so that different filesystem implementations can be swapped out.

clecocel, 2018-06-22:
Actually I thought I'd jump in here since I have found an OK solution that works well for us so far. The idea is that the bags are stored in an S3 bucket, then we have created a [Storage Gateway](https://aws.amazon.com/storagegateway/) to create a file storage that allows us to mount the S3 bucket using NFS. Then we have the bags on the filesystem.

There could other options to mount the S3 bucket ([s3fs-fuse](https://github.com/s3fs-fuse/s3fs-fuse) or [goofys](https://github.com/kahing/goofys) are candidates but we preferred going with the AWS supported method).

clecocel, 2018-06-23:
Jumping in here as I have a solution that works well for us.

The idea is to mount an S3 bucket using a [Storage Gateway](https://aws.amazon.com/storagegateway/). We then create a file storage that allows us to mount the S3 bucket using NFS on local computers, EC2 instances etc.

Other options could be to use [s3fs-fuse](https://github.com/s3fs-fuse/s3fs-fuse) or [goofys](https://github.com/kahing/goofys) to mount the bucket but we chose to use AWS' method.

pjreed, 2018-07-03:
That's pretty neat; when mounting the S3 storage, does it properly deliver "filesystem changed" events?  One problem I had encountered when using NFS to mount a remote filesystem is that there are no change notifications, which means the bag database doesn't automatically pick up new bags.

anassinator, 2018-07-04:
We use `s3fs-fuse` successfully with this and the events you mentioned seem to be working properly.

LiberiFatali, 2021-03-15:
> We use `s3fs-fuse` successfully with this and the events you mentioned seem to be working properly.

I have the issue with "filesystem changed" events. It seems Bag Database doesn't catch the event through s3fs-fuse when it's triggered in S3. Could you show a good configuration for this to work?

closed 2021-08-24 by danthony06

danthony06, 2021-08-24:
It's been quite a while, but we just released v3.3.0 that includes S3 as a storage backend.

PR #152 (2021-08-24): Refactor storage for multiple backends and add S3 support
This refactors how the Bag Database manages its bag storage.  There is now a `BagStorage` interface that different mechanisms can implement to handle different types of storage, and an arbitrary number of different storage backends can be configured in the `settings.yml` file.  Every `BagStorage` implementation has a `storageId` that is used to uniquely identify that backend and also a number of implementation-specific configuration parameters.

This also adds a `FilesystemBagStorageImpl` that provides filesystem-based storage that is b [... cut]