swri-robotics/bag-database #70: Extremely slow GPS inserts

clecocel, 2017-05-17:
We have noticed that GPS position inserts take an inordinate amount of time on our files (GPS is recorded at ~10Hz on files that can span 2 hours).

Are the inserts synchronously waiting for a response from the Postgres DB? Would it be easy to either batch insert or insert asynchronously?

pjreed, 2017-05-18:
Out of curiosity, how slow is "extremely slow"?  How long does it take to insert a 2 hour long bag file?

2 hours of GPS would be about 72,000 positions; I would expect the fastest it could insert that would be one or two minutes, probably a few times longer if it's running on a slow drive or with a slow CPU.  I haven't looked deeply into it yet, but probably the bottleneck is due to recalculating spatial indexes when inserting new positions.  I'll see if there's anything I can do to improve that.

pjreed, 2017-05-18:
For the purpose of comparison, I've been running a few tests on my local workstation.  Relevant hardware specs:

- CPU: 12-core Xeon E5-1650 @ 3.5 GHz
- Disk: 1 TB Samsung 850 EVO SSD
- RAM: 32 GB

I've tested adding a bag file that is about five hours long and has 353,805 `gps_common/GPSFix` messages in it, then made a minor tweak to it (just so the bag DB wouldn't recognize it as a duplicate file) and added it again and repeated the process a few more times.  Judging from the timestamps in the log files, each time it takes between 2.5 and 3 minutes to add all of the GPS positions to the database, which seems pretty reasonable to me.  The Bag DB's java process hovers at about 50% usage of one CPU core, occasionally spiking, and postgres hovers at about 40% of one core.

What are the hardware specs for the system your Bag DB is running on?  Also, how large is your current database?

clecocel, 2017-05-18:
A 2 hour long bag file would take around 1 hour to upload if not more (it's truly very slow). What's strange is that the CPU is mostly idle so I don't think it's a compute issue, it looks more like a network issue to me. We host the Postgres DB online and not locally, could it be a latency problem?

pjreed, 2017-05-18:
Ah, if postgres and the Bag DB are not on the same host (or at least co-located), then the network connection is probably the issue.  What's the latency and bandwidth between the hosts like?

clecocel, 2017-05-18:
I'd need to check that. That's why I was wondering if there is a way to do async inserts or at least batch them?

pjreed, 2017-05-18:
I'm not sure if it's possible to do the inserts asynchronously (or even if it's feasible to do so without causing a queue somewhere to overflow), but I'll look into it.

I'd previously taken a look at batching inserts and didn't get any performance improvement out of it, so I didn't bother with it, but it might make a difference over a slow network connection.  I'll see about enabling that and pushing a new Docker image for you to test.

clecocel, 2017-05-18:
That'd be great, thanks a lot!

clecocel, 2017-05-18:
We have approximately 30ms of roundtrip between the Postgres and the bag DB machine. 
(30 * 1e-3 * 2 * 3600 * 10) / 3600 = 0.6 hours
|__________|  |_________|  |__|
roundtrip       2 hour      10Hz

This adds up to about 0.6 hours just for the inserts. Batching may make a lot of sense for us ...

pjreed, 2017-05-18:
Just for reference, while reviewing the documentation, I also discovered that batching doesn't work at all for the identity generation strategy I was using for the `bag_positions` table.  I should be able to change it, although there will be a small change to the schema, so this won't be quite as simple as just turning on a flag...

pjreed, 2017-05-18:
Ok, I've changed how ids are generated for the `bag_positions` table and enabled batch insertions (currently with a batch size of 100), and I pushed it to a Docker image named `swrirobotics/bag-database:2.6.0-SNAPSHOT`.  It actually makes a very significant performance improvement even in my test case; the bag file that was taking 2.5 minutes to insert now takes about 40 seconds.  Give it a try and let me know how it works for you.

clecocel, 2017-05-18:
Just tested, the improvement is huge for us going from 9 minutes for a bag file containing about 17 mins worth of GPS data down to nearly instantaneous (less than 8 secs). This fix would be huge for us there's no doubt, I don't know if you plan to incorporate it in your master branch?

pjreed, 2017-05-18:
Great, I'll go ahead and merge it into devel.  I'll make a new release off of the master branch after I fix the timestamp issue.

clecocel, 2017-05-19:
That's great thanks a lot for the super quick answers and fix! I'll redeploy our bag DB as soon as you push the timestamp issue.

closed 2017-05-19 by pjreed

pjreed, 2017-05-19:
The `swrirobotics/bag-database:latest` Docker image has been updated, so you should be clear to redeploy now.

clecocel, 2017-05-19:
Great thanks again!

PR #72 (2017-05-18): Improve GPS position insertion speed
This changes how ids are generated for the bag_positions table and
enables batching in Hibernate in order to improve the speed of
inserting GPS positions.

Fixes #70