swri-robotics/bag-database #71: Wrong start and end times

clecocel, 2017-05-17:
It turns out that for many of our bag files the start and end times displayed are wrong (see below for an example). The duration is correct however for those files.

Would you know what could cause this or how to fix it?

Highlighted blue line is the one of interest
[image]
[image]

pjreed, 2017-05-18:
It finds the start & end times by iterating over all of the [ChunkInfo](http://wiki.ros.org/Bags/Format/2.0#Chunk_info) records in a bag file and selecting the earliest "start time" and the latest "end time" that it can find.  It seems really strange that it would get the times wrong but the duration right, because it calculates the duration by simply subtracting the start time from the end time.  Do you have a fairly small bag file you could give me that demonstrates the problem?

clecocel, 2017-05-18:
I can probably send you a small bag file that exhibits this behavior. What would be a good way to send it to you?

pjreed, 2017-05-18:
If it's small enough to attach it to a GitHub issue (10 MB), putting it here is fine.

clecocel, 2017-05-18:
Here's a very small fake bag that demonstrates the issue.

[test.bag.zip](https://github.com/swri-robotics/bag-database/files/1011590/test.bag.zip)

[image]

pjreed, 2017-05-18:
Good news!  The dates are being extracted from the bag file and stored in the database properly.  It's the Javascript-side renderer that's putting the wrong value in there.  Will be an easy fix.

PR #73 (2017-05-19): Display correct timestamps
The configured dateFormatter string was putting the month in the
"minute" position.  This fixes that and also consolidates all of the
date formatter strings that were scattered around into a single spot.

Fixes #71
commit c59a7e0b (2017-05-19): Fix date rendering in grids

The configured dateFormatter string was putting the month in the
"minute" position.  This fixes that and also consolidates all of the
date formatter strings that were scattered around into a single spot.

Fixes #71
closed 2017-05-19 by pjreed