swri-robotics/bag-database #78: build index for compressed rosbag

0ldWang, 2017-06-26:
hi I'm using the bag-database and it's really helpful.

Can it build index for lz4 compressed rosbags,

thx.

pjreed, 2017-06-27:
The bag database should already be able to process bags that have been compressed by running `rosbag compress --lz4` on them.  Is there something different you're trying to do?

0ldWang, 2017-06-27:
My situation is the metadata(topics, msgs) can be extracted well, but the gps info seems to be wrong. The path showed in the database web ui is only a point.(It's normal when rosbag is not compressed)

pjreed, 2017-06-27:
I see.  That should be working, so that sounds like a bug.  Do you have a small bag file you could send me that demonstrates the problem?

0ldWang, 2017-06-27:
Sure, I will send it to you tomorrow because it's a little late in Beijing time.. 

Thx for replying.

0ldWang, 2017-06-27:
By the way, I used the latest docker to run the bag-database.

0ldWang, 2017-06-28:
HI, I generated a lz4 compressed rosbag which only contains gps topic(/navsat/fix) for 5 minutes.
Have tested on my own computer that the bag-database can't generate the correct path in the web ui. 

[lz4_compressed.bag.zip](https://github.com/swri-robotics/bag-database/files/1108389/lz4_compressed.bag.zip)

pjreed, 2017-06-29:
Thanks for the bag file.  The LZ4 Java library has been updated recently, and I think this issue should be fixed in the latest development version of the bag database.  Could you try pulling the Docker image `swrirobotics/bag-database:2.7.0-SNAPSHOT` and see if that works for you?

0ldWang, 2017-06-30:
Hi

I use swrirobotics/bag-database:2.7.0-SNAPSHOT docker image to run bag-database service, it seems that another more serious error happened.

When I click on the infomation icon, it shows:
[image]

When I click the map icon, it shows:
[image]

pjreed, 2017-06-30:
Uh-oh, that's not good.  Could you try to remove and re-add that bag file, then send me the output from running `sudo docker logs bagdb`?

0ldWang, 2017-06-30:
Here is the log:

[bagdb.log.zip](https://github.com/swri-robotics/bag-database/files/1115527/bagdb.log.zip)

pjreed, 2017-06-30:
The log file looks like it successfully indexed the bag file.  Could you also try to view the details of the bag file and view the map, and then send me that log file?

0ldWang, 2017-06-30:
Here is the new log after view the map and the detail:
[bagdb.log.zip](https://github.com/swri-robotics/bag-database/files/1115685/bagdb.log.zip)

pjreed, 2017-07-01:
It looks like there's a incompatibility with the database mapping and the actual tables in the database. What version of PostgreSQL / PostGIS are you using? Have you tried this with a brand new database, or was it upgraded from an older version of the bag database?

0ldWang, 2017-07-02:
I use mdillon/postgis:latest docker to run the postgresql service.

pjreed, 2017-07-03:
Maybe something went wrong during a database upgrade; trying running this command and send me the result so I can see what your database schema looks like:

`sudo docker exec bagdb-postgres pg_dump -U bag_database -s bag_database`

pjreed, 2017-07-03:
Nevermind, I figure out what was going wrong.  It seems like at some point, columns defined with a `blob` type in a Liquibase definition began being created as `oid` columns in PostgreSQL, whereas previously they were created as `bytea` columns.  The change in columns broke the Hibernate mappings, and so any queries that touched a table with blob were failing.

I've modified the code so it will force it to create `bytea` columns, so it will work on newly-created databases.

If you can delete your database and re-create it from the beginning, please do that, then run `sudo docker pull swrirobotics/bag-database:2.7.0-SNAPSHOT` to get the latest code and try again.

pjreed, 2017-07-03:
Actually, deleting your old database won't be necessary; I pushed another update to `2.7.0-SNAPSHOT` that should be able to fix an existing database.  You should be able to just pull that and run it, and hopefully it will fix your problems.

0ldWang, 2017-07-04:
I pull the latest swrirobotics/bag-database:2.7.0-SNAPSHOT image and it works.

Thanks so much. : )

closed 2017-07-05 by pjreed