swri-robotics/bag-database #80: Host is down

0ldWang, 2017-07-04:
Hello again...

Here is my situation:
I have some bags that cannot be indexed, I can see there are many errors said "Error checking bag file: could not execute statement; SQL [n/a]; nested exception is org.hibernate.exception.DataException: could not execute statement".
And I use docker logs bags | tail, it continuously shows:
com.github.swrirobotics.bags.reader.exceptions.BagReaderException: java.nio.file.FileSystemException: /bags/xxx.bag: Host is down
	at com.github.swrirobotics.bags.reader.BagFile.forMessagesOnTopic(BagFile.java:346)
	at com.github.swrirobotics.bags.BagService.getAllGpsMessages(BagService.java:1083)
	at com.github.swrirobotics.bags.BagService.updateBagFile(BagService.java:1534)
	at com.github.swrirobotics.bags.filesystem.BagScanner$FullScanner.lambda$0(BagScanner.java:441)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.nio.file.FileSystemException: /bags/xxx.bag: Host is down
	at sun.nio.fs.UnixException.translateToIOException(UnixException.java:91)
	at sun.nio.fs.UnixException.rethrowAsIOException(UnixException.java:102)
	at sun.nio.fs.UnixException.rethrowAsIOException(UnixException.java:107)
	at sun.nio.fs.UnixFileSystemProvider.newFileChannel(UnixFileSystemProvider.java:177)
	at java.nio.channels.FileChannel.open(FileChannel.java:287)
	at java.nio.channels.FileChannel.open(FileChannel.java:335)
	at com.github.swrirobotics.bags.reader.BagFile.getChannel(BagFile.java:127)
	at com.github.swrirobotics.bags.reader.BagFile.forMessagesOnTopic(BagFile.java:335)

I'm wondering if it will call a service but the service is down?

0ldWang, 2017-07-04:
BTW, The rosbags are mounted from another machine.

pjreed, 2017-07-05:
It seems like it had an error connecting to the other machine.  How is your bag directory mounted? (sshfs / smb / nfs ...)  Can you try using something like `rosbag info` to read the bag files on the machine that bagdb is running on to see if you can read them from there?

0ldWang, 2017-07-06:
the directory is mounted with cifs.
rosbag info command works fine.
BTW, some bags are indexed successfully except the gps information. Metadata looks good, but the map icon neither be transparent that cannot be clicked, nor the gps path not completed.

0ldWang, 2017-07-10:
update:

It seems like the exception is caused by the mounted fs, maybe the network is not stable so reading the rosbag will throw "Host is down" exception.

I will close this issue.

Thanks @pjreed

closed 2017-07-10 by 0ldWang

pjreed, 2017-07-10:
Yeah, I think that network instability could cause it to have problems reading the files.

By the way, if the map icon for a bag is transparent, that means it was unable to find any GPS coordinates in that bag file; that could happen if they're on a topic that is not listed in the `GPS_TOPICS` in your Docker environment variable.