swri-robotics/bag-database #81: Manually adding vehicle name

fabolhak, 2017-10-08:
Hey, 
I was wondering if it is possible to manually add vehicle names and location? This is supported by the UI (both fields are editable and there is a save button). However after editing, pressing save and reloading the page, the entered vehicle name and location is gone. 

[image]

I'm using a docker installation with BagDatabase 2.6.

pjreed, 2017-10-09:
That *should* be working, and I know it worked at some time, but I just tested it and it's not working locally for me, either.  I'll investigate.

pjreed, 2017-10-09:
Strangely, I saw the problem once, but after restarting my Docker containers the problem went away.  Does it still happen for you after restarting them?  Could you upload the bag file from your `bagdb` container after reproducing it?

fabolhak, 2017-10-09:
Thanks for the fast reply. 
> Does it still happen for you after restarting them?
I did try to restart the `bagdb` container several times and editing the file, however without any luck :(

> Could you upload the bag file from your bagdb container after reproducing it?
I uploaded one *.bag file where I tried to edit the vehicle name [here](https://ufile.io/l6d8z)

Is this information stored inside the bag-files? 

The bag files are stored in a subfolder structure. I changed permissions to 777 to all the folders and files. However, still not working.

pjreed, 2017-10-09:
The bag database never actually modifies the bag files; it's actually possible to mount the Docker volume containing your bags as read only and it will work fine.

All of the editable fields are stored in the `bags` table in the Postgres database, so there may be an issue writing to the database; watch the log for it like so:
```
sudo docker logs --tail=30 -f bagdb
```

Then try to modify one of the fields and see if it logs an error of some sort.

fabolhak, 2017-10-09:
```
2017-10-09 16:25:46.647 [http-nio-8080-exec-9] INFO  c.g.s.bags.BagController - getBag: 8
2017-10-09 16:25:46.875 [http-nio-8080-exec-9] DEBUG c.g.swrirobotics.bags.BagService - Successfully got bag: drive_2017-10-05-14-26-04_homography_1.bag
2017-10-09 16:25:51.377 [http-nio-8080-exec-6] INFO  c.g.s.bags.BagController - updateBags
2017-10-09 16:25:51.546 [http-nio-8080-exec-6] INFO  c.g.s.bags.BagController - Processing 1 bags.
2017-10-09 16:25:51.603 [http-nio-8080-exec-6] WARN  c.g.s.bags.BagController - Unexpected error updating bags:
java.lang.NullPointerException: null
	at com.github.swrirobotics.bags.BagService.makePoint(BagService.java:883)
	at com.github.swrirobotics.bags.BagService.updateBag_aroundBody8(BagService.java:890)
	at com.github.swrirobotics.bags.BagService$AjcClosure9.run(BagService.java:1)
	at org.springframework.transaction.aspectj.AbstractTransactionAspect.ajc$around$org_springframework_transaction_aspectj_AbstractTransactionAspect$1$2a73e96cproceed(AbstractTransactionAspect.aj:66)
	at org.springframework.transaction.aspectj.AbstractTransactionAspect$AbstractTransactionAspect$1.proceedWithInvocation(AbstractTransactionAspect.aj:72)
	at org.springframework.transaction.interceptor.TransactionAspectSupport.invokeWithinTransaction(TransactionAspectSupport.java:281)
	at org.springframework.transaction.aspectj.AbstractTransactionAspect.ajc$around$org_springframework_transaction_aspectj_AbstractTransactionAspect$1$2a73e96c(AbstractTransactionAspect.aj:70)
	at com.github.swrirobotics.bags.BagService.updateBag(BagService.java:887)
	at com.github.swrirobotics.bags.BagController.updateBags(BagController.java:159)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.springframework.web.method.support.InvocableHandlerMethod.doInvoke(InvocableHandlerMethod.java:221)
	at org.springframework.web.method.support.InvocableHandlerMethod.invokeForRequest(InvocableHandlerMethod.java:136)
	at org.springframework.web.servlet.mvc.method.annotation.ServletInvocableHandlerMethod.invokeAndHandle(ServletInvocableHandlerMethod.java:114)
	at org.springframework.web.servlet.mvc.method.annotation.RequestMappingHandlerAdapter.invokeHandlerMethod(RequestMappingHandlerAdapter.java:827)
	at org.springframework.web.servlet.mvc.method.annotation.RequestMappingHandlerAdapter.handleInternal(RequestMappingHandlerAdapter.java:738)
	at org.springframework.web.servlet.mvc.method.AbstractHandlerMethodAdapter.handle(AbstractHandlerMethodAdapter.java:85)
	at org.springframework.web.servlet.DispatcherServlet.doDispatch(DispatcherServlet.java:963)
	at org.springframework.web.servlet.DispatcherServlet.doService(DispatcherServlet.java:897)
	at org.springframework.web.servlet.FrameworkServlet.processRequest(FrameworkServlet.java:970)
	at org.springframework.web.servlet.FrameworkServlet.doPost(FrameworkServlet.java:872)
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:661)
	at org.springframework.web.servlet.FrameworkServlet.service(FrameworkServlet.java:846)
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:742)
	at org.apache.catalina.core.ApplicationFilterChain.internalDoFilter(ApplicationFilterChain.java:231)
	at org.apache.catalina.core.ApplicationFilterChain.doFilter(ApplicationFilterChain.java:166)
	at org.springframework.web.filter.CharacterEncodingFilter.doFilterInternal(CharacterEncodingFilter.java:197)
	at org.springframework.web.filter.OncePerRequestFilter.doFilter(OncePerRequestFilter.java:107)
	at org.apache.catalina.core.ApplicationFilterChain.internalDoFilter(ApplicationFilterChain.java:193)
	at org.apache.catalina.core.ApplicationFilterChain.doFilter(ApplicationFilterChain.java:166)
	at org.apache.tomcat.websocket.server.WsFilter.doFilter(WsFilter.java:52)
	at org.apache.catalina.core.ApplicationFilterChain.internalDoFilter(ApplicationFilterChain.java:193)
	at org.apache.catalina.core.ApplicationFilterChain.doFilter(ApplicationFilterChain.java:166)
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:316)
	at org.springframework.security.web.access.intercept.FilterSecurityInterceptor.invoke(FilterSecurityInterceptor.java:126)
	at org.springframework.security.web.access.intercept.FilterSecurityInterceptor.doFilter(FilterSecurityInterceptor.java:90)
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:330)
	at org.springframework.security.web.access.ExceptionTranslationFilter.doFilter(ExceptionTranslationFilter.java:114)
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:330)
	at org.springframework.security.web.session.SessionManagementFilter.doFilter(SessionManagementFilter.java:122)
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:330)
	at org.springframework.security.web.authentication.AnonymousAuthenticationFilter.doFilter(AnonymousAuthenticationFilter.java:111)
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:330)
	at org.springframework.security.web.servletapi.SecurityContextHolderAwareRequestFilter.doFilter(SecurityContextHolderAwareRequestFilter.java:169)
[... cut ...]
	at org.apache.coyote.AbstractProtocol$ConnectionHandler.process(AbstractProtocol.java:861)
	at org.apache.tomcat.util.net.NioEndpoint$SocketProcessor.doRun(NioEndpoint.java:1455)
	at org.apache.tomcat.util.net.SocketProcessorBase.run(SocketProcessorBase.java:49)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at org.apache.tomcat.util.threads.TaskThread$WrappingRunnable.run(TaskThread.java:61)
	at java.lang.Thread.run(Thread.java:748)
```

pjreed, 2017-10-09:
Aha, it looks like the code that updates the metadata in the database is assuming the bag will have at least one GPS coordinate in it, and that exception is happening for bags that don't have any.  Should be easy to fix.

pjreed, 2017-10-09:
I think I've fixed the issue; try pulling the `swrirobotics/bag-database:2.7.0-SNAPSHOT` Docker image and see if that works for you.

fabolhak, 2017-10-09:
Thank you for this quick fix. It works for me :) 👍

PR #82 (2017-10-11): Fix setting bag metadata
Manually setting metadata would fail for bag files that did not have
any GPS coordinates.

Fixes #81
commit ff9ce96d (2017-10-11): Fix setting bag metadata

Manually setting metadata would fail for bag files that did not have
any GPS coordinates.

Fixes #81
closed 2017-10-09 by fabolhak