swri-robotics/bag-database #92: Adding tags to Bags

joelsa, 2018-06-07:
I tried to add some tags to bags I recorded, as stated in the features list:

> Tagging: Bags can be tagged and searched for with arbitrary metadata strings. Existing tags on arbitrary metadata topics in bag files will be automatically read.

I started the bag database using:
` -e METADATA_TOPICS="/metadata" \`
on the `docker run` command, made sure `rosbag record` subscribed to **/metadata** and then typed:
`rostopic pub /metadata std_msgs/String "name: John Doe"`. I have also tried both leading and trailing newlines.

The message was recorded in my bag and can be seen using rqt_graph. But when I copy the bag file into the Bag Database, the tags are not shown. The Interface correctly shows that _something_ has been published to /metadata but the tags column on the right hand side doesn't list "name: John Doe" and I am unable to find anything searching for this tag.

Is there an error on my side during the process or is it even supposed to work that way?

pjreed, 2018-07-03:
For what it's worth, it _should_ work that way.  I've been busy lately and haven't had a chance to test it, but I'll see if I can repeat that sometime soon...

cheffe112, 2018-07-24:
Well, the Readme says:

> The messages on the topic should be newline-separated tags that are made of colon-separated key/value pairs; for example:
> name: John Doe
> email: jdoe@example.com

How I understood this is that `name: John Doe` is the key and `email: jdoe@example.com` is the value, so what you @joelsa  want to achieve should follow a `name\nJohn Doe` syntax, I guess?

However, for me it doesn't work either this way while specifying `METADATA_TOPICS="/annotation"` with the `/annotation` topic existing as std_msgs/String and containing some messages.

joelsa, 2018-08-29:
That's certainly one way to read it @cheffe112, however by looking at the code I am quite sure that the pairs should be of the format` tagName + ": " + value`

tafulop, 2019-06-18:
I have the same problem and I would like to help you to fix it. To do that, I need information about what is the Java version that should be used to build the solution. I am trying to do it with version 1.8.0_151, but it is not okay unfortunately.

pjreed, 2019-06-18:
Everything is designed to build with Java 1.8.  What errors are you encountering?

tafulop, 2019-06-18:
Same behaviour that was mentioned in this bug, the tags are not get parsed, even the content of the bag file is correct. I am looking into the problem, right now I am debugging the code and hopefully will find the problem. :)

tafulop, 2019-06-18:
I have found the problem, the array that meant to contain the metadata topics is not set anywhere, therefore stays empty, therefore skips the tag reading. I have manually set the topic to the default one by hardcoding it - just to prove my idea - and it worked, so the parsing and persisting functionality works for the tags.

Shall I fix it?

pjreed, 2019-06-18:
Sure, please submit a pull request.

tafulop, 2019-06-18:
Alright, I will do in the next few days.

pjreed, 2019-07-11:
Well, it took a long time to figure out, but this should finally be fixed.

closed 2019-07-11 by pjreed

tafulop, 2019-07-18:
Something is still not okay, I think the config file is not generated by the script that you have just added but it exists beforehand - at least it looks like this with the latest version of development branch.