swri-robotics/bag-database #97: Bag files folder is not scanning.

fnc11, 2019-07-03:
Dear Developers, 
I tried to imitate the project on my machine, Initially, I used JDK 11 as it was mentioned that we can use JDK 8.0 or higher, but then I ran into the below error:
Could not find artifact com.sun:tools:jar:11.0.3 at specified path /usr/lib/jvm/java-11-openjdk-amd64
Some [answers](https://stackoverflow.com/questions/8375423/missing-artifact-com-suntoolsjar) suggested to add an exclusion in the dependency tree, but this didn't solve my problem.

Tools.jar has been deprecated after Java 8, so no longer supported.
And since exclusion methods failed so I decided to fall back to Java 8 version.

After that everything built smoothly. But now I ran into some different issue, the interface is looking as it should, but the BagScanner is not scanning files in the specified directory(/bags) 
`2019-07-03 16:22:59.998 [pool-2-thread-1] INFO  c.g.s.bags.filesystem.BagScanner - Scanning for new bag files in /bags.
2019-07-03 16:22:59.999 [pool-2-thread-1] DEBUG c.g.s.bags.filesystem.BagScanner -   Hey I am inside the getBagFiles method
2019-07-03 16:23:00.007 [pool-2-thread-1] DEBUG c.g.s.bags.filesystem.BagScanner -   Hey I am inside the file reader: /bags
2019-07-03 16:23:00.008 [pool-2-thread-1] DEBUG c.g.s.bags.filesystem.BagScanner -   Now printing by method 1: 
2019-07-03 16:23:00.010 [pool-2-thread-1] DEBUG c.g.s.bags.filesystem.BagScanner - Found 0 bag files on disk.
2019-07-03 16:23:00.019 [pool-2-thread-1] DEBUG c.g.swrirobotics.bags.BagService - Scanning bags already in the database.
2019-07-03 16:23:00.020 [pool-2-thread-1] DEBUG c.g.s.bags.filesystem.BagScanner - 0 bags exist in the database and are not missing.
2019-07-03 16:23:00.021 [pool-2-thread-1] DEBUG c.g.s.bags.filesystem.BagScanner - 0 in the DB are missing on disk.
`
and if I try with different directory it gives this error:
`2019-07-03 16:23:49.342 [pool-2-thread-1] DEBUG c.g.s.bags.filesystem.BagScanner - Done checking bag files.
2019-07-03 16:23:55.268 [http-nio-8080-exec-22] INFO  c.g.s.bags.filesystem.BagScanner - Scheduling a quick directory scan.
2019-07-03 16:23:55.270 [pool-2-thread-1] INFO  c.g.s.bags.filesystem.BagScanner - Scanning for new bag files in /home/fnc11/Hiwi/IKA/testBag.
2019-07-03 16:23:55.271 [pool-2-thread-1] DEBUG c.g.s.bags.filesystem.BagScanner -   Hey I am inside the getBagFiles method
2019-07-03 16:23:55.272 [pool-2-thread-1] ERROR c.g.s.bags.filesystem.BagScanner - Error parsing directory:
java.nio.file.NoSuchFileException: /home/fnc11/Hiwi/IKA/testBag
	at sun.nio.fs.UnixException.translateToIOException(UnixException.java:86)
	at sun.nio.fs.UnixException.rethrowAsIOException(UnixException.java:102)
	at sun.nio.fs.UnixException.rethrowAsIOException(UnixException.java:107)
	at sun.nio.fs.UnixFileSystemProvider.newDirectoryStream(UnixFileSystemProvider.java:427)
	at java.nio.file.Files.newDirectoryStream(Files.java:525)
	at com.github.swrirobotics.bags.filesystem.BagScanner.getBagFiles(BagScanner.java:368)
	at com.github.swrirobotics.bags.filesystem.BagScanner$FullScanner.run(BagScanner.java:428)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
2019-07-03 16:23:55.274 [pool-2-thread-1] DEBUG c.g.s.bags.filesystem.BagScanner -   Now printing by method 1: 
2019-07-03 16:23:55.276 [pool-2-thread-1] WARN  c.g.s.bags.filesystem.BagScanner - Unexpected exception when checking bag files: 
java.lang.NullPointerException: null
	at com.github.swrirobotics.bags.filesystem.BagScanner.getBagFiles(BagScanner.java:381)
	at com.github.swrirobotics.bags.filesystem.BagScanner$FullScanner.run(BagScanner.java:428)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
`

And these are the different versions of the technologies that I am using:
`03-Jul-2019 16:19:03.166 INFO [main] org.apache.catalina.startup.VersionLoggerListener.log Server version:        Apache Tomcat/9.0.2
03-Jul-2019 16:19:03.166 INFO [main] org.apache.catalina.startup.VersionLoggerListener.log Server built:          Nov 25 2017 21:08:02 UTC
03-Jul-2019 16:19:03.166 INFO [main] org.apache.catalina.startup.VersionLoggerListener.log Server number:         9.0.2.0
03-Jul-2019 16:19:03.166 INFO [main] org.apache.catalina.startup.VersionLoggerListener.log OS Name:               Linux
03-Jul-2019 16:19:03.166 INFO [main] org.apache.catalina.startup.VersionLoggerListener.log OS Version:            4.15.0-54-generic
03-Jul-2019 16:19:03.167 INFO [main] org.apache.catalina.startup.VersionLoggerListener.log Architecture:          amd64
03-Jul-2019 16:19:03.167 INFO [main] org.apache.catalina.startup.VersionLoggerListener.log Java Home:             /usr/lib/jvm/java-1.8-openjdk/jre
03-Jul-2019 16:19:03.167 INFO [main] org.apache.catalina.startup.VersionLoggerListener.log JVM Version:           1.8.0_151-b12
03-Jul-2019 16:19:03.167 INFO [main] org.apache.catalina.startup.VersionLoggerListener.log JVM Vendor:            Oracle Corporation
03-Jul-2019 16:19:03.167 INFO [main] org.apache.catalina.startup.VersionLoggerListener.log CATALINA_BASE:         /usr/local/tomcat
03-Jul-2019 16:19:03.167 INFO [main] org.apache.catalina.startup.VersionLoggerListener.log CATALINA_HOME:         /usr/local/tomcat
`

pjreed, 2019-07-10:
Hm... how are you running it?  Is this inside a Docker container?  If so, what do your volume mappings look like?

fnc11, 2019-07-11:
Yeah, I am running it inside a docker container. I am kind of novice in the area of docker so I don't know that much about the different kind of settings that we have in a docker or in a container.
Nevertheless, I have attached the output of 
`docker inspect bagdb` 
[inpect_bagdb.txt](https://github.com/swri-robotics/bag-database/files/3381564/inpect_bagdb.txt)
`docker inspect bagdb-postgres'
[inpect_bagdb_postgres.txt](https://github.com/swri-robotics/bag-database/files/3381563/inpect_bagdb_postgres.txt)

`docker volume ls` gave the below output:
`fnc11@fnc11:~$ docker volume ls
DRIVER              VOLUME NAME
local               7e14613f7497e82cdb4ae799e8f714a0da523982b10437d8887f01e21281e2eb
local               dfe896383d3aa90c8f0f4f810df33798302749a649f0fcdfba1c5e5c7da921e7
`
Let me know if you want some other details, and also guide me in which direction I should search for the solution.
I tried with both by building the image and also pulling it from docker hub then running it, both had the same problem.

pjreed, 2019-07-11:
What parameters are you using when you create/run the Docker container?  I suspect that you might not be mapping your directory containing your bag files to `/bags/` inside the container.  In this example:

```bash
docker run -d \
    -p 8080:8080 \
    -v /bag/location:/bags \
    --name bagdb \
    --net bagdb \
    -e DB_DRIVER=org.postgresql.Driver \
    -e DB_PASS=letmein \
    -e DB_URL="jdbc:postgresql://bagdb-postgres/bag_database" \
    -e DB_USER=bag_database \
    -e METADATA_TOPICS="/metadata" \
    -e VEHICLE_NAME_TOPICS="/vehicle_name" \
    -e GPS_TOPICS="/localization/gps, /gps, /imu/fix" \
    swrirobotics/bag-database:latest
```

The ` -v /bag/location:/bags` argument means that the path on the host filesystem at `/bag/location` will be mapped to `/bags` inside the container, which is where the database will scan for files by default; if it's not mapped properly, then `/bags` inside the container may be empty.

(By the way, I just pushed a new version yesterday that fixes and issue with parsing some of the environment variables, so I recommend pulling the latest image)

fnc11, 2019-07-11:
I am running the container as specified in the repository, tried with the latest image as well. These are the logs from the container bagdb.
[logs_bagdb.txt](https://github.com/swri-robotics/bag-database/files/3383229/logs_bagdb.txt)

As you can see there is a file in /bags folder.
[image]

pjreed, 2019-07-11:
The first part of the argument, `/bag/location`, is the location on your host where the bag directory is; so if they're in `/bags` on the host, try `-v /bags:/bags` instead and see if that behaves differently.

fnc11, 2019-07-11:
Yes, thank you very much.
This solved my problem, I thought /bags is the location where I have to put bag files and /bag/location is some kind of parameter which takes this location as input.
Thanks again, I was stuck in this from last 2 weeks.

fnc11, 2019-07-11:
One more thing is there any documentation kind of thing for this project, which we can use to understand the functioning of the different files how the whole project structured and what input/output goes where that would be absolutely helpful.

closed 2019-07-11 by fnc11

pjreed, 2019-07-11:
The project uses the [Spring](https://spring.io/projects) library fairly heavily; if you are familiar with Spring MVC, the way the code is organized into Controllers / Services / Repositories should be fairly straightforward.  If you're not familiar with it, it is a pretty heavyweight library and can take some work to get the hang of it, but here is a decent introduction to it: https://spring.io/guides/tutorials/rest/

With that said, the Controller classes define the REST API for the web interface uses, and their publicly exposed methods should have Javadocs describing how they work; if you wanted to programmatically access the API to search for or manipulate bags, for example, that's all in the BagController: https://github.com/swri-robotics/bag-database/blob/master/src/main/java/com/github/swrirobotics/bags/BagController.java