swri-robotics/bag-database #128: Add finer-grained access controls

pjreed, 2020-11-06:
Currently, the Bag DB operates in basically one of two modes:
1. User authentication is not required, in which case any user can upload files, scripts, and modify mutable data
2. LDAP authentication is required, in which case users must log in, and after logging in can do any of those things

In both cases, a user must authentication with an admin password to perform administrative tasks.

It would be nice to have more control over exactly what authenticated and unauthenticated users are allowed to do.  A good first goal would be to be able to selectively control whether unauthenticated users can upload files, scripts, or modify data.

As a more long-term goal, it would be useful to have a set of configurable user roles that can be assigned to users; off the top of my head, possible permissions would include:
1. Upload bag files
2. Modify bag metadata
3. Modify scripts
4. Run scripts
5. Access administrative functions

ptulpen, 2020-11-20:
Hello,
additionally it would be great to have the access controlled by ldap groups 
(maybe just the access/authentication by that and the roles/authorization by internal methods )

What also could help there is either supporting kerberos SSO directly or using the remote_user variable from an apache or nginx 

I know, big points to a wishlist but they help a lot in a bigger environment

(I would like to help you with this, but I have no java skills at all)

PhilippSchmaelzle, 2024-07-01:
I would like to add something to this wishlist as well :)
Just started to use bag_database and it is awesome!

What I see as a useful extension to LDAP would be restricted access to the bags themself. 
e.g. there are uncritical bag files which can be accessed by all users. But there might be also bag files containing image data which are GDPR (General Data Protection Regulation) relevant, so not everybody should have access to those.
So only users within a certain LDAP group are allowed to view the contained images or download the bags. 
(I could imagine, that the metadata are visible to all, as they are uncritical. So everybody can see, that possible useful data is available, but he/she just do not have the correct permissions to continue)

Do you see such a LDAP group access feature on your roadmap?