swri-robotics/bag-database #129: ROS2 support?

Georacer, 2020-11-13:
Hello everyone!

I might be mistaken, but ROS2 bag files have a different format than ROS1.

Does bag-database support ROS2 bag files?

Thank you!

pjreed, 2020-11-13:
You are correct, ROS2 bag files have a different format; in fact, rosbag2 is designed to have a plugin-based storage backend, so it is not technically tied to any specific format at all, although the default implementation stores data as SQLite databases.

The Bag Database doesn't currently support ROS2 bags, although I'd like to add support for them in the future.

In order support ROS2 bags, it would need a Java library that is capable of reading them; just reading SQLite databases is easy, but things get a little bit more complicated because the default rosbag2 implementation serializes messages using the [DDS cdr](https://github.com/ros2/rosbag2#serialization-format-plugin-architecture) format, so extracting useful information from the messages would also require being able to deserialize that.

Georacer, 2020-11-13:
I see... thank you for your answer! Cheers!

Timple, 2021-07-16:
Not sure if it helps, but there is already some ROS2 - java combining going on:
https://github.com/ros2-java/ros2_java

danthony06, 2021-08-24:
There's been some movement on making ros2 bags fully self describing, which would help out this issue quite a bit. See the discussion on this issue: https://github.com/ros2/rosbag2/issues/782

pjreed, 2021-08-24:
Oh that'd be nice.  I was just looking into this again the other day, actually...

Basically, there are two problems right now standing in the way of this:
1. ROS2 bags don't contain their message definitions (unlike ROS1 bags).  This means it's impossible to parse messages without having message definitions.  Hopefully the linked issue will result in a solution for that.  In the meantime, if we want to be able to ingest ROS2 bags, we may have to just accept only being able to parse a subset of known message types.
2. The message data inserted into the standard SQL format is, as mentioned previously, in DDS CDR format and needs to be deserialized in order to get anything useful out of it.  I had briefly hoped that maybe this was identical to CORBA CDR format and tried using JacORB (a pure-Java CORBA implementation) to parse it, but had a bunch of problems.  It's probably possible to write our own parser, but it'd be a pain.

It might be possible to incorporate ros2-java into it in order to use Fast DDS to deserialize messages... while it'd be nice to have a pure Java implementation that doesn't rely on having a whole ROS stack included in the image to function, it's probably worth investigating to get a feel for how difficult that would be to use vs. writing our own CDR deserializer.

danthony06, 2021-08-24:
@pjreed what if as part of the Docker image building process, we cloned the standard ROS message definitions, and optionally user defined repos, built the message definitions/CDR representations, and then found a Java library for deserializing CDR formatted messages? We could then use the message definitions we built in the first stage and for deserializing information in the database. That way, we wouldn't have a runtime dependency on ROS but could still get some level of introspection.

pjreed, 2021-08-24:
Yep.  I have specifically had enough experiences with ACE/TAO in the past that I'm shy about touching anything that uses it.  I do see that they provide a pre-built Docker image, so maybe it would be possible to do something like make our own image based on that which can take in message definitions, serialized data, and output it in some other format we can easily interpret... but that might be just as much work as making our own CDR parser.

pjreed, 2021-08-24:
Just thought I'd point out that, while it's not useful for ingesting things on the backend, here's a Typescript-based CDR implementation and ROS2 message deserializer:
https://github.com/foxglove/cdr
https://github.com/foxglove/rosmsg2-serialization

Might be useful as a reference for writing our own.

danthony06, 2021-09-02:
I'm hoping with some of the conversations that have occurred in the Tooling
WG that this might be resolved in the near future. I'll push on it again in
the next WG meeting, and see if we can get more traction.

danthony06, 2021-11-02:
It sounds like the H-release of ROS2 will have the ability to store message definitions in the bag file, which should make it more feasible to implement ROS 2 support.

gjaeger1, 2022-09-29:
Hi all, 

I'm currently working on a project that sets out to acquire large datasets (3D LiDAR, camera data, GNSS, etc.) of outdoor environments. We would like to use bag-database for that, but are using ROS2. Are there any news on whether/when bag-database will be able to handle ROS2 bags as well?

From the looks of it, this project could be very helpful in maintaining a consistent overview on our ros bags and their data!

danthony06, 2022-09-29:
@gjaeger1 It might be worth revisiting this. ROS2 bag support still will be tricky. Support for storing message definitions in an MCAP file, not a bag file, was recently added: https://github.com/ros2/rosbag2/issues/782. Unfortunately, embedding those message definitions in an MCAP file still requires having those message definitions available on the system doing the recording. So unlike ROS1, you can't attach a computer with a minimal ROS installation to your robot and have it read the message definitions off the wire for storage in a ROS bag.

That being said, adding support for MCAP at this point seems reasonable. There's a bunch of different languages supported for it (https://github.com/foxglove/mcap), so we may be able to just pull off of their native interface to add it to the bag database.

smash0190, 2024-05-03:
Just wanted to see if there are any new developments with regards to mcap support?

danthony06, 2024-05-03:
It looks like we'll have to write our own MCAP decoder or wrap one of the
existing implementations. Let me look at this a little bit over the
weekend, because I was thinking about it today.

smash0190, 2024-05-23:
Did you get a chance to look?

space192, 2024-07-10:
any updates on the support of mcap ?

shrijitsingh99, 2024-08-22:
This is something I am interested in too, an potentially can contribute some cycles to help get up and running. If you have some idea of what would be a good path forward, I can start looking into it