swri-robotics/bag-database #134: The upload function shows Unknown when the filesize is bigger than 7GB

hzung, 2021-03-05:
### Steps to reproduce
Step 0. Download this example bag file (10GB): [http://infinity.csail.mit.edu/data/2011/2011-01-27-07-40-55.bag](http://infinity.csail.mit.edu/data/2011/2011-01-27-07-40-55.bag)
Step 1. Login into the dashboard using **admin/letmein** account.
Step 2. Click the **Upload Bags** button.
Step 3. Browse to the file **2011-01-27-07-40-55.bag**
Step 4. Click the **Upload All** button.
Step 5. Observe the result.

[image]

Is there any workaround solution for this case?

pjreed, 2021-03-07:
Thanks for the report.  I looked into it and it looks like by default, the server is configured to only accept a maximum upload size of 4GB.  I'll increase the maximum upload size to 50 GB, which should be good enough for the vast majority of bag files.

It'd be better to make this a configurable setting, but I experimented with it for a while and couldn't figure out a way to actually make the value configurable; the code that configures the maximum upload size in the server is run before the component that reads the configuration file is loaded... but this should be good enough for now.

hzung, 2021-03-10:
@pjreed Thanks, man. This works like a charm ❤

PR #135 (2021-03-10): Increase max upload size to 50GB
The maximum upload size was set to 4GB, which is smaller than many bag files, so this increases the maximum to 50GB.  It also displays a more useful error message in the upload box if the size is exceeded.

Fixes #134
commit c09cb2bf (2021-03-10): Increase max upload size to 50GB

The maximum upload size was set to 4GB, which is smaller than many bag files, so this increases the maximum to 50GB.  It also displays a more useful error message in the upload box if the size is exceeded.

Fixes #134