swri-robotics/bag-database #139: Unsupported image encoding: bgra8

phil333, 2021-03-14:
Hi, thank you for this amazing tool! Could you add support for bgra8 image topics. The error I get says:

```
Error retrieving image:
com.github.swrirobotics.bags.reader.exceptions.BagReaderException: Unsupported image encoding: bgra8
```

This addition would cover popular Zed stereo camera images to be covered as well! If you can

On a sidenote, another suggestion would be to add graph plotting for numerical values, which would be handy to quickly assess a bag.

danthony06, 2021-03-22:
I've made a PR to support this here: https://github.com/swri-robotics/bag-database/pull/141. Is it possible for you to supply a small bag file from the Zed camera so I can test it?

phil333, 2021-03-22:
yes, will do so tomorrow

phil333, 2021-03-24:
Here is a small bag, containing image and compressed image of both color and depth images from the zed2 camera
https://wetransfer.com/downloads/2c0f5a909a99724f06aa2bb0d0c18db220210324103725/a2915a64f7af41e369b4e72fb662b55f20210324103816/36aca1

pjreed, 2021-04-05:
Thanks for the report, this should be fixed now.

closed 2021-04-05 by pjreed