swri-robotics/bag-database #140: How to allow redirecting ldap_login with more custom parameters?

hzung, 2021-03-16:
When I try to redirect the `ldap_login` to a new error such as `/ldap_login?error_captcha`. The server keeps automatically redirect this URL to `/ldap_login`. The system seems to allow only redirect to `/ldap_login?error`.

Is there anyone who knows where to control this redirection rule?

I try to find all possible places I can think of, but no luck.
Thank you so much.

pjreed, 2021-03-29:
When LDAP is enabled, the page used for handling logins is defined at https://github.com/swri-robotics/bag-database/blob/master/src/main/java/com/github/swrirobotics/config/SecurityConfig.java#L159 .  Any unauthenticated request will be directed to that page.

The controller that handles requests to `ldap_login` is very simple: https://github.com/swri-robotics/bag-database/blob/master/src/main/java/com/github/swrirobotics/ldap/LdapLoginController.java

All it does is use the MVC view `ldap/ldap_login`.  The HTML template for that view is here: https://github.com/swri-robotics/bag-database/blob/master/src/main/webapp/WEB-INF/views/ldap/ldap_login.html

Requesting `ldap_login?error_captcha` should result in a parameter named `error_captcha` being set in that view, which you could handle similar to the existing `error` and `logout` parameters to change how the page appears.

hzung, 2021-04-07:
Thank @pjreed. I will try it.

closed 2021-04-07 by hzung