swri-robotics/bag-database #147: Can't get running script with Docker-Compose "Without Authentication"

Thehkart, 2021-05-14:
So i recently got into dockers and bag-files because of my work. We were looking for a way to store Bag files and this system was very promissing. But i ran into the Problem that i can't get Scripts to work in my docker. 

[image]

Upper script is used to test if everything works fine. But I get this error Message:
[image]

It is also to mention that I am using Windows 10 as my OS.
Because of that I had to change the path for the Deamon in the Registry version into following:
- ./daemon.json:/ProgramData/docker/config/daemon.json 

I tried this because i hoped to get it running with this version because there was a Docker Image set to use in the Script Properties.

I hope there is some help because I'm going crazy as I don't understand how to solve this issue.

pjreed, 2021-05-14:
I think that the error is probably because you are using this as the docker image: `swrirobotics/bag-database:8080/ros:melodic`

Docker expects that everything before the first `/` is a valid hostname and everything after that is an image tag; so in this case, it would be trying to connect to a host name `swrirobotics` and getting an image tagged with `bag-database:8080/ros:melodic`, but that is not a valid tag name, and so Docker is returning the error message `invalid reference format`.

To make sure everything is working, try first using an image just named `ros:melodic`, which will pull it from the official Docker hub.  If you need to host docker images on a private registry, check the "With a Private Registry" page in the documentation for information on how to configure that.

Thehkart, 2021-05-14:
First thanks for your reply and the explanation of the error :)

Oh yeah when I run ros:melodic as a container and use it as the docker:image everything is working fine for the example script.

If I use the docker-compose.yml with: ./daemon.json:/ProgramData/docker/config/daemon.json

and use the command "docker-compose up" i get following:
[image]

Has it something to do with Windows or am I missing on something important?

pjreed, 2021-05-14:
Actually, I think that might be a problem in the example configuration.  The `registry` service expects to find a volume named `registry`; try modifying the `volumes:` section at the bottom of your docker-compose.yml file and add a line that says `registry:` to create a shared volume with that name.

Thehkart, 2021-05-15:
Ok this seems to work. 
Also the example script is working fine when I use "ros:melodic" as the docker image unlike in the example with "registry:5000/ros:melodic".

If I use it like the example i get following output in the logs:
[image]

like it's unable to connect to the host "registry". But I did change the deamon file and I am referencing it like in the comment before.

Thehkart, 2021-05-17:
So a little update to my problem.
I don't get the timeout anymore. But I am unable to use the registry:5000/ros:melodic as a docker image.
The resulting error image looks something like this:
[image]

I already updated the insecure-registries in my docker gui:
[image]

Also in the local daemon.json
[image]

and in the docker-compose.yml I got following:
[image]

but apparently my "dockerd" command isn't running. 
I try to use it in my powershell but I only get error messages

pjreed, 2021-05-17:
That looks reasonable; could you upload the contents of your `docker-compose.yml` file and also the contents of the log file from when you run `docker-compose up`?

Thehkart, 2021-05-18:
The `docker-compose.yml` looks like:
version: '3.6'
services:
    docker:
        image: docker:dind
        privileged: yes
        networks:
            - bagdb
        volumes:
            - bags:/bags:ro
            - scripts:/scripts
            - docker_cache:/var/lib/docker
            - ./daemon.json:/ProgramData/docker/config/daemon.json
        command: ["dockerd", "--host=tcp://0.0.0.0:2375"]
        # SSL is disabled since only the Bag Database can access this anyway.
        # Be careful about allowing anything else to access this service! 
    registry:
        image: registry:2
        networks: 
            - bagdb
        ports:
            - "5000:5000"
        volumes:
            - registry:/var/lib/registry 
    bagdb:
        image: swrirobotics/bag-database:latest
        networks:
            - bagdb
        depends_on:
            - postgres
        ports:
            - "8080:8080"
        volumes:
            - bags:/bags
            - indexes:/root/.ros-bag-database/indexes
            - scripts:/scripts
        environment:
            ADMIN_PASSWORD: "letmein"  # Change this to something more secure
            DB_DRIVER: org.postgresql.Driver
            DB_PASS: letmein  # Should match POSTGRES_PASSWORD below
            DB_URL: "jdbc:postgresql://postgres/bag_database"  # Should reference POSTGRES_DB below
            DB_USER: bag_database  # Should match POSTGRES_USER below
            DOCKER_HOST: "http://docker:2375"
            GPS_TOPICS: "/localization/gps, gps, /vehicle/gps/fix, /localization/sensors/gps/novatel/raw, /localization/sensors/gps/novatel/fix, /imu_3dm_node/gps/fix, /local_xy_origin"  # Add topics where you publish GPS coordinates
            METADATA_TOPICS: "/metadata"
            VEHICLE_NAME_TOPICS: "/vms/vehicle_name, /vehicle_name"  # Replace with a topic on which you publish your vehicle's name
    postgres:
        image: postgis/postgis:11-2.5
        networks:
            - bagdb
        volumes:
            - postgres:/var/lib/postgresql/data
        ports:
            - "5432:5432"
            # This port is exposed to make it easy for you to connect to the database with a
            # SQL client to perform operations on it.   If you don't need to do so, this
            # port does not need to be exposed.
        environment:
            POSTGRES_PASSWORD: letmein # If you do expose it, it's a good idea to change this password to something more secure.
            POSTGRES_USER: bag_database
            POSTGRES_DB: bag_database
networks:
    bagdb: {}
volumes:
    bags:
        driver: local
        driver_opts:
            type: 'none'
            o: 'bind'
            device: '/var/local/bags'  # Replace this with the bath to your bags
    docker_cache:
    postgres:
    ldap:
    slapd:
    indexes:
    registry:
    scripts:
        driver_opts:
            type: 'tmpfs'
            device: 'tmpfs'

And the logs after running `docker-compose up`. I hope this helpts

[image]
[image]
[image]
[image]
[image]

pjreed, 2021-05-19:
One thing that sticks out to me as being a little strange here is I don't see any entries in the log file for the `registry_1` container, and I would expect it to print something even if it failed.  Does `docker ps -a` list a container named `bag-database_registry_1`?  If so, what does it print when you run `docker logs bag-database_registry_1`?

Thehkart, 2021-05-19:
Yes. The registry is listed
For the question how the logs look like:

[image]