swri-robotics/bag-database #151: "Open with" feature or Compatibility with webviz

RapIt, 2021-07-13:
Hi,
I was looking into some additional visualization options when I came across [Webviz](https://github.com/cruise-automation/webviz/). While I really like your image and video preview feature, most of our bags have only pcl data. So I was wondering if webviz would be a good addition to our current setup with only the bag-database.

Ideally I would want to click an "open with" button in bagdb generating a customizable link in the form of `https://domain.org/webviz/?remote-bag-url=https://domain.org/bagdb/bags/download?id=1` 
Maybe a generic user-definable "open with" functionality could do that.

To play the bagfile, webviz would require the bagdb webserver to have cors enabled and to be able to accept range requests, which is not possible atm. I could also solve this using a third small webserver component (open_with -> webserver -> api_call for bag path info from id -> redirect to webviz).

Did someone already try to achieve something similar?
Thanks!

danthony06, 2021-07-14:
@pjreed what are your thoughts? I haven't tried this before, but it seems like it would take a moderate amount of effort, especially if we are concerned about cross-platform compatibility. 

Could the "Run Script" functionality accomplish what you want @RapIt? Could you run a script that would launch your process and pass the bag into it?

RapIt, 2021-07-14:
I think the script functionality is not applicable here. To clarify, I actually meant two different things:

1. "Open With": Similar to the context menu option "Copy link" there could be another entry like "Open Link with...". I think it would require some work to make this link generation generic and supporting templated bag attributes (so that the user has a setting where the link can be composed, eg. `LINK=https://domain.org/webviz/ + <BAG_PATH_ID>`) Of course, for my purpose a single hardcoded link generation would be sufficient. We could do that link generation with a very simple container as "script" but this wouldn't integrate into the UI.

2. Enable Cors and Range requests: To make webviz able to stream a bagfile from a "remote server", they require this server to allow range requests (see. https://github.com/cruise-automation/webviz/issues/247). We could fire up another minimal webserver  fulfilling these requirements and which could serve the bags that way. 
Of course it would be cleaner to have only Bag-database running which is already able to serve the bags but the cors and range requests are not supported atm. I think it would be a small change but it may raise security concerns and one would have to expose the origin settings, etc. Maybe serving large bags over your http endpoint is not fast enough for webviz and we would need a direct file handle after all, but this is would be an issue on webviz side.

pjreed, 2021-07-15:
I haven't had a chance to look into it much yet, but I suspect that support for this would actually be pretty easy to add.  Spring has some built-in mechanisms for configuring [CORS](https://www.baeldung.com/spring-cors) and handling [Range](https://melgenek.github.io/spring-video-service#responding-with-parts-of-data) requests, they probably just need to be added to the appropriate methods for retrieving bag files.  The hardest part will probably be just setting up another server that can test it...

RapIt, 2021-07-15:
Yes this looks fairly simple indeed. As for development testing one could just fire up webviz `docker run -p 8080:8080 cruise/webviz` and then point to `localhost:8080/?remote-bag-url=http://localhost:9090/bagdb/bags/download?bagId=1` to see if it starts streaming the bag. For simple (or automated) inspection you can also just get the http headers with `curl` and see if the attributes are set.

pjreed, 2021-08-24:
Just thought I'd mention here that I've made some progress on this in https://github.com/hatchbed/bag-database/tree/feature/cors-and-range-support .  I've added CORS and Range support to the `/bags/download` API call, and I can copy & paste the URL to download a bag into webviz on my local browser and it works fine.  I'll still need to do some GUI work to add someting like an "Open with..." option into the UI and make it configurable.

closed 2021-08-30 by danthony06

frapit, 2021-08-31:
Great news, thank you for implementing it. It is exactly what I was looking for. I just deployed it and it seems to work perfectly.

PR #155 (2021-08-30): Support integrating with external applications
This adds CORS and Range support to the `/bags/download` API call, which means it is now possible for web applications hosted on a different domain to download ranges of bytes from bag files.  This makes it possible to visualized bags stored in the Bag Database directly in applications like [Webviz](https://webviz.io) or  [Foxglove Studio](https://foxglove.dev/).

Additionally, it is possible to provide a list of such external applications in the Bag Database's configuration, and it will present them as options in the context-menu that appears whe [... cut]