swri-robotics/bag-database #153: Network Error while playing a video

UniBwTAS, 2021-08-16:
Hello,
thank you for your nice tool. We tried it and is works well. However, when we try to play a video from an image topic we get the following exception after few seconds. Can you tell us whats is wrong here?

```
bagdb_1     | 2021-08-16 18:10:42.233 [http-nio-8080-exec-9] INFO  c.g.s.bags.BagController - getBag: 16
bagdb_1     | 2021-08-16 18:10:42.254 [http-nio-8080-exec-9] DEBUG c.g.swrirobotics.bags.BagService - Successfully got bag: 2021-08-11-14-47-56.bag
bagdb_1     | 2021-08-16 18:10:48.256 [http-nio-8080-exec-19] INFO  c.g.s.bags.BagController - getVideo: 16:/sensor/camera/marveye/left/image_raw
bagdb_1     | 2021-08-16 18:10:48.257 [http-nio-8080-exec-19] INFO  c.g.s.bags.BagController - Finished getVideo()
bagdb_1     | 2021-08-16 18:10:48.639 [mvcTaskExecutor-6] DEBUG c.g.swrirobotics.bags.BagService - Expecting 4356 frames.
bagdb_1     | 2021-08-16 18:10:48.639 [mvcTaskExecutor-6] DEBUG c.g.swrirobotics.bags.BagService - Reading message from bag 16 on topic [/sensor/camera/marveye/left/image_raw]
bagdb_1     | 2021-08-16 18:10:50.007 [mvcTaskExecutor-6] INFO  c.g.swrirobotics.bags.BagService - Starting video stream.
bagdb_1     | 2021-08-16 18:10:50.017 [mvcTaskExecutor-6] DEBUG c.g.swrirobotics.bags.BagService - Image format: bayer_bggr8 / 2448x900 / 435.6s / 10.0 Hz
bagdb_1     | 2021-08-16 18:10:50.021 [mvcTaskExecutor-6] INFO  c.g.swrirobotics.bags.BagService - Beginning to stream image data to ffmpeg.
bagdb_1     | 2021-08-16 18:10:50.021 [Thread-14] DEBUG c.g.swrirobotics.bags.BagService - Piping data from ffmpeg to the client.
bagdb_1     | 2021-08-16 18:11:19.029 [mvcTaskExecutor-6] WARN  c.g.s.b.r.r.ChunkRecordIterator - Error reading data chunk
bagdb_1     | com.github.swrirobotics.bags.reader.exceptions.BagReaderException: java.nio.channels.ClosedByInterruptException
bagdb_1     | 	at com.github.swrirobotics.bags.reader.records.Record.readData(Record.java:192)
bagdb_1     | 	at com.github.swrirobotics.bags.reader.records.ChunkRecordIterator.findNext(ChunkRecordIterator.java:118)
bagdb_1     | 	at com.github.swrirobotics.bags.reader.records.ChunkRecordIterator.hasNext(ChunkRecordIterator.java:78)
bagdb_1     | 	at com.github.swrirobotics.bags.reader.messages.serialization.MsgIterator.findNext(MsgIterator.java:144)
bagdb_1     | 	at com.github.swrirobotics.bags.reader.messages.serialization.MsgIterator.findNext(MsgIterator.java:206)
bagdb_1     | 	at com.github.swrirobotics.bags.reader.messages.serialization.MsgIterator.hasNext(MsgIterator.java:111)
bagdb_1     | 	at com.github.swrirobotics.bags.reader.BagFile.forMessagesOnTopic(BagFile.java:394)
bagdb_1     | 	at com.github.swrirobotics.bags.BagService.writeVideoStream(BagService.java:726)
bagdb_1     | 	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
bagdb_1     | 	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
bagdb_1     | 	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
bagdb_1     | 	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
bagdb_1     | 	at org.springframework.aop.support.AopUtils.invokeJoinpointUsingReflection(AopUtils.java:343)
bagdb_1     | 	at org.springframework.aop.framework.ReflectiveMethodInvocation.invokeJoinpoint(ReflectiveMethodInvocation.java:197)
bagdb_1     | 	at org.springframework.aop.framework.CglibAopProxy$CglibMethodInvocation.invokeJoinpoint(CglibAopProxy.java:749)
bagdb_1     | 	at org.springframework.aop.framework.ReflectiveMethodInvocation.proceed(ReflectiveMethodInvocation.java:163)
bagdb_1     | 	at org.springframework.aop.framework.CglibAopProxy$DynamicAdvisedInterceptor.intercept(CglibAopProxy.java:688)
bagdb_1     | 	at com.github.swrirobotics.bags.BagService$$EnhancerBySpringCGLIB$$c72927a0.writeVideoStream(<generated>)
bagdb_1     | 	at com.github.swrirobotics.bags.BagController.lambda$getVideo$0(BagController.java:142)
bagdb_1     | 	at org.springframework.web.servlet.mvc.method.annotation.StreamingResponseBodyReturnValueHandler$StreamingResponseBodyTask.call(StreamingResponseBodyReturnValueHandler.java:110)
bagdb_1     | 	at org.springframework.web.servlet.mvc.method.annotation.StreamingResponseBodyReturnValueHandler$StreamingResponseBodyTask.call(StreamingResponseBodyReturnValueHandler.java:97)
bagdb_1     | 	at org.springframework.web.context.request.async.WebAsyncManager.lambda$startCallableProcessing$4(WebAsyncManager.java:326)
bagdb_1     | 	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:515)
bagdb_1     | 	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
bagdb_1     | 	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
bagdb_1     | 	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
bagdb_1     | 	at java.base/java.lang.Thread.run(Thread.java:829)
bagdb_1     | Caused by: java.nio.channels.ClosedByInterruptException: null
bagdb_1     | 	at java.base/java.nio.channels.spi.AbstractInterruptibleChannel.end(AbstractInterruptibleChannel.java:199)
bagdb_1     | 	at java.base/sun.nio.ch.FileChannelImpl.endBlocking(FileChannelImpl.java:162)
bagdb_1     | 	at java.base/sun.nio.ch.FileChannelImpl.read(FileChannelImpl.java:228)
bagdb_1     | 	at com.github.swrirobotics.bags.reader.records.Record.readData(Record.java:155)
bagdb_1     | 	... 26 common frames omitted
bagdb_1     | 2021-08-16 18:11:19.029 [mvcTaskExecutor-6] WARN  c.g.swrirobotics.bags.BagService - Interrupted waiting for consumer to finish.
bagdb_1     | 2021-08-16 18:11:19.029 [mvcTaskExecutor-6] INFO  c.g.swrirobotics.bags.BagService - Done streaming video.
bagdb_1     | 16-Aug-2021 18:11:19.036 WARNING [http-nio-8080-exec-4] org.apache.catalina.core.AsyncContextImpl.setErrorState onError() call failed for listener of type [org.apache.catalina.core.AsyncListenerWrapper]
bagdb_1     | 	java.lang.IllegalArgumentException: Cannot dispatch without an AsyncContext
bagdb_1     | 		at org.springframework.util.Assert.notNull(Assert.java:193)
bagdb_1     | 		at org.springframework.web.context.request.async.StandardServletAsyncWebRequest.dispatch(StandardServletAsyncWebRequest.java:131)
bagdb_1     | 		at org.springframework.web.context.request.async.WebAsyncManager.setConcurrentResultAndDispatch(WebAsyncManager.java:389)
bagdb_1     | 		at org.springframework.web.context.request.async.WebAsyncManager.lambda$startCallableProcessing$2(WebAsyncManager.java:313)
bagdb_1     | 		at org.springframework.web.context.request.async.StandardServletAsyncWebRequest.lambda$onError$0(StandardServletAsyncWebRequest.java:146)
bagdb_1     | 		at java.base/java.util.ArrayList.forEach(ArrayList.java:1541)
bagdb_1     | 		at org.springframework.web.context.request.async.StandardServletAsyncWebRequest.onError(StandardServletAsyncWebRequest.java:146)
bagdb_1     | 		at org.apache.catalina.core.AsyncListenerWrapper.fireOnError(AsyncListenerWrapper.java:49)
bagdb_1     | 		at org.apache.catalina.core.AsyncContextImpl.setErrorState(AsyncContextImpl.java:413)
bagdb_1     | 		at org.apache.catalina.connector.CoyoteAdapter.asyncDispatch(CoyoteAdapter.java:253)
bagdb_1     | 		at org.apache.coyote.AbstractProcessor.dispatch(AbstractProcessor.java:241)
bagdb_1     | 		at org.apache.coyote.AbstractProcessorLight.process(AbstractProcessorLight.java:59)
bagdb_1     | 		at org.apache.coyote.AbstractProtocol$ConnectionHandler.process(AbstractProtocol.java:893)
bagdb_1     | 		at org.apache.tomcat.util.net.NioEndpoint$SocketProcessor.doRun(NioEndpoint.java:1723)
bagdb_1     | 		at org.apache.tomcat.util.net.SocketProcessorBase.run(SocketProcessorBase.java:49)
bagdb_1     | 		at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
bagdb_1     | 		at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
bagdb_1     | 		at org.apache.tomcat.util.threads.TaskThread$WrappingRunnable.run(TaskThread.java:61)
bagdb_1     | 		at java.base/java.lang.Thread.run(Thread.java:829)
bagdb_1     | 2021-08-16 18:11:19.049 [Thread-14] ERROR c.g.swrirobotics.bags.BagService - Error processing ffmpeg output:
bagdb_1     | org.apache.coyote.CloseNowException: Failed write
bagdb_1     | 	at org.apache.coyote.http11.Http11OutputBuffer$SocketOutputBuffer.doWrite(Http11OutputBuffer.java:549)
[... cut ...]
bagdb_1     | 	at org.apache.commons.io.IOUtils.copyLarge(IOUtils.java:1128)
bagdb_1     | 	at org.apache.commons.io.IOUtils.copy(IOUtils.java:849)
bagdb_1     | 	at org.apache.commons.io.IOUtils.copyLarge(IOUtils.java:1104)
bagdb_1     | 	at org.apache.commons.io.IOUtils.copy(IOUtils.java:825)
bagdb_1     | 	at com.github.swrirobotics.bags.BagService$FfmpegImageHandler$OutputConsumer.run(BagService.java:335)
bagdb_1     | 2021-08-16 18:11:19.049 [Thread-14] DEBUG c.g.swrirobotics.bags.BagService - Finished processing output from ffmpeg.
```

danthony06, 2021-08-16:
@pjreed looks like it's encoded as bayer_bggr8. Could that be causing issues?

AndreasR30, 2021-08-16:
The video starts playing for about 15 seconds. Then the error occurs. So I dont think it has something to do with the encoding.

pjreed, 2021-08-16:
Hmm... it looks like this is the exception that is the root of the issue:
```
bagdb_1     | 2021-08-16 18:11:19.049 [Thread-14] ERROR c.g.swrirobotics.bags.BagService - Error processing ffmpeg output:
bagdb_1     | org.apache.coyote.CloseNowException: Failed write
bagdb_1     | 	at org.apache.coyote.http11.Http11OutputBuffer$SocketOutputBuffer.doWrite(Http11OutputBuffer.java:549)
```

I would expect it to fail to write to the HTTP stream if the web browser closes the connection, but it's unusual that would happen while the video is streaming unless they do something like close their web browser, and earlier it says the video is 435.6s long, so it shouldn't be finishing that fast.

@danthony06 , do you have a bag file with a long-duration video that you could test it with?  Or @AndreasR30 , would it be possible for you to share a bag file that reproduces the issue?

danthony06, 2021-08-16:
I don't have one handy, I could just make a fake one though.

AndreasR30, 2021-08-16:
Our Bag files are quite big (60GB). I will try to generate a smaller one.

danthony06, 2021-08-16:
I think we should be able to make a fake one if necessary.

AndreasR30, 2021-08-17:
Here is a bag file, which causes Problems in our bag database:
https://1drv.ms/u/s!AjmdCIatIC7FgYdQuleMg8cMVrCz-A?e=wXULAJ

AndreasR30, 2021-08-17:
Maybe it has something to do with large bag files because with smaller ones the problem did not occur.

pjreed, 2021-08-17:
Maybe.  I just tried your bag file and can confirm I see the issue on my local computer, although the video runs for about 30 seconds before it has an exception.  I'll keep looking into it and see if I figure out what could be causing it.

AndreasR30, 2021-08-17:
@pjreed Thank you very much.

danthony06, 2021-08-17:
https://github.com/swri-robotics/bag-database/pull/154 should have fixed this. You can build from source, or we should be pushing a new release in the next few days.

closed 2021-08-17 by danthony06

AndreasR30, 2021-08-19:
Wow that was fast. We tested it and it works very well now. Thank you very much!

danthony06, 2021-08-24:
v3.3.0 was just released and includes this fix.

PR #154 (2021-08-17): Extend async task timeout
Spring was killing async tasks that ran longer than 30 seconds, which was preventing users from streaming long videos.  This extends the timeout to 1 hour.

Fixes #153.