swri-robotics/bag-database #173: getting error 401 when uploading

ptulpen, 2022-02-16:
Hello, 
I try to get uploads working via curl. 

CSRF=$(${curlcmd} -s -L -X GET    ${URL} | sed -rn 's/^.*(value=|csrfToken = )"([-a-z0-9]*)".*$/\2/p')  || { echo "${URL} not accessible" ; exit 1; }
CSRF=$(${curlcmd} -s   -F "username=admin" -F "password=${bagpassword}" -F "_csrf=${CSRF}"   -L ${URL}/signin | sed -rn 's/^.*csrfToken = "([-a-z0-9]*)".*$/\1/p')
curl --key ${clientkey} --cert  ${clientcertificate} -c ${cookiefile}  -b ${cookiefile}
 -X POST -H "Content-Type: multipart/form-data"  -F targetDirectory=test  -F file=@test.bag  ${URL}/bags/upload

Gut what I in the access log from the bagserver is
"POST /bag/bags/upload HTTP/1.1" 401 713
are there any special parameters needed?

danthony06, 2022-02-17:
Can you create any files in the upload directory? Perhaps try something like ```touch /path/to/bags/test.txt```?

ptulpen, 2022-02-17:
Hi, the writing from inside the container works

ptulpen, 2022-02-21:
I found to issue myself :  
since I need different authentication methods for interactive usage and api upload, I created different locations in the apache proxy. But only the one for interactive login I had the correct BAG_DB Path set.
The 401 was only misleading 

The solution was now to combine the authentication methods for the same location as the BAG_DB path

closed 2022-02-21 by ptulpen