swri-robotics/bag-database #190: Working on extending the db - need help with Sencha

xxxZer0xxx, 2022-05-09:
Hey, I am working on extending the DB to be able to upload CSV Files to separate View. However I am struggling to work with the ID I get when clicking in the main view to transfer it to the "detailed Window" view. Could use a little help here.

And I was wondering if I can use plain JS within in the framework to work with arrays from postgres.

Regards

pjreed, 2022-05-09:
For an example of how that works, you might take a look at how the bag grid creates details windows? In the `showBagDetails` function at https://github.com/swri-robotics/bag-database/blob/master/src/main/webapp/resources/js/views/BagGrid.js#L632, it creates a new object by using the `bagDetailsWindow` xtype (defined at https://github.com/swri-robotics/bag-database/blob/master/src/main/webapp/resources/js/views/BagDetailsWindow.js#L33), passes in the bag ID as an argument which is available as a property (as seen https://github.com/swri-robotics/bag-database/blob/master/src/main/webapp/resources/js/views/BagDetailsWindow.js#L88 ), and then shows the window.

> And I was wondering if I can use plain JS within in the framework to work with arrays from postgres.

You can, but it might be a lot plainer than you're expecting since it doesn't use any modern tools like npm or webpack for building the application.  External Javascript libraries it uses are stored in the file tree under https://github.com/swri-robotics/bag-database/tree/master/src/main/webapp/resources/js/external, then there's a step in the maven build where it minifies and combines everything here: https://github.com/swri-robotics/bag-database/blob/master/pom.xml#L531

Refactoring it to use modern tooling has been on my list of things I'd like to do for a long time now, I just haven't gotten around to it.  I'd really prefer to replace Sencha with something like Vue / Vuetify, too...

xxxZer0xxx, 2022-05-09:
Thanks a lot! Gonna try it out.

Already got another Question, is it possible to test the Spring Controller implementation with an external App like Postman, when I am using the "No Authentication" Method?

pjreed, 2022-05-09:
Yep, it should be pretty straightforward.  I looked around a little bit and found this: https://github.com/fbenz/restdocs-to-postman

I was able to build the bag database (`mvn package`), then use that tool to convert the output of the Spring REST Docs in the `target` directory, import that into Postman, and I was able to make API calls without doing any kind of authentication.

xxxZer0xxx, 2022-05-09:
Nice, thanks again! :)

danthony06, 2022-05-22:
Closing this as it seems resolved. Please open up more issues if you have other questions.

closed 2022-05-22 by danthony06