swri-robotics/bag-database #209: Unauthorized error using docker compose

gigcastro, 2024-03-13:
Hi,

I'm attempting to install bagdb using the Without Authentication method.

Tried doing docker compose up and got:
```
 ✘ bagdb Error                                                                                                     1.2s 
Error response from daemon: Head "https://ghcr.io/v2/swri-robotics/bag-database/manifests/latest": unauthorized
```
Also tried with docker run directly and got:
```
...
Status: Downloaded newer image for ghcr.io/swri-robotics/bag-database/bag-database:latest
37ee247f1154c221dc89234852474c31b98d74f2db98722027d95fd9851cc467
docker: Error response from daemon: network bagdb not found.
```

Is there something I should do to being able to pull from the docker repositories?

danthony06, 2024-03-13:
Are you logged into ghcr.io?

gigcastro, 2024-03-13:
I wasn't correctly logged in, you where right.
But now I'm getting:
`Error response from daemon: denied`

I might've some missing permission in the token right?

broccoli-smuggler, 2024-03-25:
I also have the same issue. Did anyone find a fix? Logged in to ghcr.io using standard login and trying to run docker-compose up.

broccoli-smuggler, 2024-03-25:
Changing the docker-compose.yaml file to use `swrirobotics/bag-database:latest` instead seems to allow pulling the image.

danthony06, 2024-04-30:
@broccoli-smuggler I'm having a hard time reproducing this. If I run

`docker pull ghcr.io/swri-robotics/bag-database:latest`

I can successfully pull the latest image. Which docker file did you end up changing to fix the problem on your end?

danthony06, 2024-04-30:
Okay, I think some of the problems may be because we had a lot of docker registry paths that looked like `swri-robotics/bag-database/bag-database:tag` (note the duplicated bag-database), and we partially switched over to paths like `swri-robotics/bag-database:tag`. I just updated the documentation and Docker files to the simplified paths, pushed and tagged the updated version, and released v3.5.3 that has all of these changes.